syssi/esphome-ant-bms #1: Collect BMS responses

syssi, 2022-02-20:
@lucasimons I've pushed a skeleton of a new custom component to this repository. Could you assemble a test setup using `esp32-example.yaml` and attach your BMS? This implementation sends the following message/request periodically to the BMS:

```
[21:21:44][D][uart_debug:114]: >>> 5A:5A:00:00:01:01
```

The BMS should respond with `<<<`. Could you collect and provide some responses? I will add a response parser later on.

https://github.com/syssi/esphome-jk-bms/discussions/38

lucasimons, 2022-02-20:
ok I'm preparing the setup to do the tests with my 300a bms,  I have attached the project already started, I think it is useful to then adapt and integrate it on esphome

[antbms.zip](https://github.com/syssi/esphome-ant-bms/files/8104939/antbms.zip)

syssi, 2022-02-20:
Your implementation helps a lot. As soon as some responses are available I'm able to do some testes and provide a first sensor example. You could contribute additional sensors if you like.

syssi, 2022-02-21:
@lucasimons I've prepared a pull request to illustrate how to implement/add new sensors: https://github.com/syssi/esphome-ant-bms/pull/6/files

May be you like to contribute some.

lucasimons, 2022-02-21:
[image]

syssi, 2022-02-21:
The traffic doesn't look good to me. The BMS should respond to requests only. Every response should start with `<<< AA:55:AA:FF:...`

lucasimons, 2022-02-21:
now i m using this script with a raspy

#!/usr/bin/env python
import sys
import time
import serial
import struct
from binascii import unhexlify
import requests as req
import bluetooth
from bluetooth import *
import sys

#if sys.version < '3':
#    input = raw_input

bd_addr = "AA:BB:CC:AA:23:54"

port = 1

#sock=bluetooth.BluetoothSocket( bluetooth.RFCOMM )
#sock.connect((bd_addr, port))
#sock.close()
#time.sleep(1)

#Define RS485 serial port
ser = serial.Serial(
    port='/dev/rfcomm0',
    baudrate = 9600,
    parity=serial.PARITY_NONE,
    stopbits=serial.STOPBITS_ONE,
    bytesize=serial.EIGHTBITS,
    timeout = 1)

import paho.mqtt.client as mqtt

def on_message(client, userdata, msg):
    print(msg.topic)

def on_connect(client, userdata, flags, rc):
    print("Connected! Result code: " + str(rc))
    client.subscribe("$SYS/#")

client = mqtt.Client("bms")
client.on_connect = on_connect
client.on_message = on_message
#set up username and password for your MQTT server
client.username_pw_set("mosquitto", password="pass")
#ip address/website of MQTT server
client.connect("192.168.X.XXX", 1883, 60)
client.loop_start()

#104
MOSFET_Discharge_St=["OFF","ON","cell overdischarge","overcurrent","4","pack overdischarge",
"bat overtemp","MOSFET overtemp","Abnormal current","battery is not detected",
"PCB overtemp","charge MOSFET turn on","shortcircuit","Discharge MOSFET abnormality",
"Start exception","Manual off"]
#103
MOSFET_Charge_St=["OFF","ON","overcharge","overcurrent","batt full","pack overvoltage",
"bat overtemp","MOSFET overtemp","abnormal current","bat not detected",
"PCB overtemp","11-undefined","12-undefined","Discharge MOSFET abnormality","14","Manual off"]
#134-135
Bal_St=["OFF","limit trigger exceeds","charging v diff too high","overtemp","ACTIVE",
        "5-udef","6-udef","7-udef","8-udef","9-udef","PCB Overtemp"]

while True :
 test='DBDB00000000'
 try:
  ser.write (test.decode('hex'))
 except:
[... 214 lines cut ...]
   cell_6 = struct.unpack('>H',unhexlify(cell_6))[0]*0.001
   print cell_6
   client.publish("battery/cell_6", cell_6)
 except:
   pass
 #Cell_7
 try:
   cell_7 = (Antw33.encode('hex') [(18*2):(19*2+2)])
   cell_7 = struct.unpack('>H',unhexlify(cell_7))[0]*0.001
   print cell_7
   client.publish("battery/cell_7", cell_7)
 except:
   pass
 #Cell_8
 try:
   cell_8 = (Antw33.encode('hex') [(20*2):(21*2+2)])
   cell_8 = struct.unpack('>H',unhexlify(cell_8))[0]*0.001
   print cell_8
   client.publish("battery/cell_8", cell_8)
 except:
   pass

 time.sleep(2)

ser.close()

lucasimons, 2022-02-23:
I had reversed some cables now I get this data[image]

syssi, 2022-02-23:
Could you copy&paste the log messages here? I don't want to write off the screenshots. :-)

lucasimons, 2022-02-23:
[17:39:35][D][uart_debug:114]: >>> 5A:5A:00:00:01:01
[17:39:35][D][uart_debug:114]: <<< 63:7B:73:21:69:63:63:31:FD:21:85:21:21:DC:33:23:AD:21:35:31:A1:69:3B:20:63:00:6B:61:63:23:63:32:FF:63:7B:69:71:8D:21:21:21:2C:A4:63:7B:5B:21:69:41:62:61:2B:FF
[17:39:36][D][uart_debug:114]: <<< 63:7B:5B:21:29:61:62:61:63:43:6B:61:31:61:63:61:63:23:63:63:63:63:63:23:2A:63:CC:23:6A:61:63:16:F9
[17:39:36][D][uart_debug:114]: <<< 63:7B:5B:21:29:61:42:61:63:63:43:61:63:63:63:61:63:43:43:41:63:63:F4
[17:39:36][D][uart_debug:114]: <<< 63:7B:5B:21:29:61:42:61:63:63:63:61:63:63:63:61:63:FD
[17:39:36][D][uart_debug:114]: <<< 63:7B:5B:21:29:61:42:61:63:63:63:61:63:63:6A:FF
[17:39:36][D][uart_debug:114]: <<< 63:7B:5B:21:29:61:42:61:63:63:63:61:43:4B:FF
[17:39:36][D][uart_debug:114]: <<< 63:7B:5B:21:29:61:42:61:63:63:63:61:42:FD
[17:39:37][D][uart_debug:114]: <<< 63:7B:5B:21:29:61:42:61:63:63:63:61:CF:FF
[17:39:37][D][uart_debug:114]: <<< 63:7B:5B:21:29:61:42:61:63:63:63:41:CF:FF
[17:39:37][D][uart_debug:114]: <<< 63:7B:5B:21:29:61:42:61:63:63:63:41:CF:FF
[17:39:37][D][uart_debug:114]: <<< 63:7B:5B:21:29:61:42:61:63:63:63:41:CF:FF
[17:39:37][D][uart_debug:114]: <<< 63:7B:5B:21:29:61:42:61:63:63:63:41:CF:FF
[17:39:37][D][uart_debug:114]: <<< 63:7B:5B:21:29:61:42:61:63:63:63:41:CF:FF
[17:39:37][D][uart_debug:114]: <<< 63:7B:5B:21:29:61:42:61:63:63:63:41:CF:FF
[17:39:38][D][uart_debug:114]: <<< 63:7B:5B:21:29:61:42:61:63:63:63:41:CF:FF
[17:39:38][D][uart_debug:114]: <<< 63:7B:5B:21:29:61:42:61:63:63:63:41:CF:FF
[17:39:38][D][uart_debug:114]: <<< 63:7B:5B:21:29:61:42:61:63:63:63:41:CF:FF
[17:39:38][D][uart_debug:114]: <<< 63:7B:5B:21:29:61:42:61:63:63:63:41:CF:FF
[17:39:38][D][uart_debug:114]: <<< 63:7B:5B:21:29:61:42:61:63:63:63:41:CF:FF
[17:39:38][D][uart_debug:114]: <<< 63:7B:5B:21:29:61:42:61:63:63:63:41:CF:FF
[17:39:38][D][uart_debug:114]: <<< 63:7B:5B:21:29:61:42:61:63:63:63:41:CF:FF
[17:39:39][D][uart_debug:114]: <<< 63:7B:5B:21:29:61:42:61:63:63:63:41:CF:FF
[17:39:39][D][uart_debug:114]: <<< 63:7B:5B:21:29:61:42:61:63:63:63:41:CF:FF
[17:39:39][D][uart_debug:114]: <<< 63:7B:5B:21:29:61:42:61:63:63:63:41:CF:FF
[17:39:39][D][uart_debug:114]: <<< 63:7B:5B:21:29:61:42:61:63:63:63:41:CF:FF
[17:39:39][D][uart_debug:114]: <<< 63:7B:5B:21:29:61:42:61:63:63:63:41:CF:FF
[17:39:39][D][uart_debug:114]: <<< 63:7B:5B:21:29:61:42:61:63:63:63:41:CF:FF
[17:39:39][D][uart_debug:114]: <<< 63:7B:5B:21:29:61:42:61:63:63:63:41:CF:FF
[17:39:40][D][uart_debug:114]: <<< 63:7B:5B:21:29:61:42:61:63:63:63:41:CF:FF
[17:39:40][D][uart_debug:114]: <<< 63:7B:5B:21:29:61:42:61:63:63:63:41:CF:F
[logs_ant-bms_logs.txt](https://github.com/syssi/esphome-ant-bms/files/8126135/logs_ant-bms_logs.txt)
F

syssi, 2022-02-23:
The UART traffic looks like garbage. On every request (all 2 seconds):

```
>>> 5A:5A:00:00:01:01
```

The BMS should respond (directly) with:

```
<<< AA:55:AA:FF:02:....... 
```

After this response there must be 2 seconds **silence**. If you see `[D][uart_debug:114]: <<< ...` permanently something is wrong. First step is to disconnect the BMS. If the BMS is disconnect you should see the `>>> Requests` only and *no* incoming (`<<<`) UART traffic. 

Please provide some photos of your setup. I need to know the used GPIOs and connected pins at the BMS.

lucasimons, 2022-02-23:
[image][image]

syssi, 2022-02-23:
Could you provide the complete yaml configuration? If you use the `logger` component you aren't allowed to use `RX` and `TX`  because the logger already writes there.

lucasimons, 2022-02-23:
ok so what pin should i connect to?

this is my config

logger:
  level: VERY_VERBOSE
  logs:
    api.service: WARN
    ota: WARN
    sensor: WARN
    wif: WARN

uart:
  id: uart0
  baud_rate: 19200
  rx_buffer_size: 384
  tx_pin: RX
  rx_pin: TX
  debug:
    direction: BOTH

ant_modbus:
  id: modbus0
  uart_id: uart0

ant_bms:
  id: bms0
  ant_modbus_id: modbus0

sensor:
  - platform: ant_bms
    capacity_remaining:
      name: "${name} capacity remaining"

syssi, 2022-02-23:
`GPIO16` and `GPIO17` are safe:

```
uart:
  id: uart0
  baud_rate: 19200
  rx_buffer_size: 384
  tx_pin: GPIO16
  rx_pin: GPIO17
  debug:
    direction: BOTH
```

syssi, 2022-02-23:
`GPIO16` and `GPIO4` is also a good choice.

lucasimons, 2022-02-23:
ok now i think that it work i see the bms responce
[logs_ant-bms_logs (2).txt](https://github.com/syssi/esphome-ant-bms/files/8126501/logs_ant-bms_logs.2.txt)
e

syssi, 2022-02-23:
Perfect! This is a valid response!

lucasimons, 2022-02-23:
I try to use a custom component of @SeByDocKy and it works fine 

[image]

syssi, 2022-02-23:
If I push your response data

```
AA:55:AA:FF:01:06:0C:CF:0C:C9:0C:CB:0C:CF:0C:CF:0C:D0:0C:CF:0C:C3:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:02:FA:F0:80:00:0B:EB:C2:00:00:00:03:00:00:E8:28:00:0E:00:0D:00:0C:00:0A:FF:D8:FF:D8:00:00:00:00:00:00:00:00:00:00:27:0F:06:0C:D0:10:0C:C3:0C:CC:08:00:09:00:00:00:00:02:11:00:00:00:00:00:00:12:D3
```

into this custom component I receive:

```
[sensor:125]: 'ant-bms total voltage': Sending state 26.20000 V with 2 decimals of accuracy
[sensor:125]: 'ant-bms cell voltage 1': Sending state 3.27900 V with 3 decimals of accuracy
[sensor:125]: 'ant-bms cell voltage 2': Sending state 3.27300 V with 3 decimals of accuracy
[sensor:125]: 'ant-bms cell voltage 3': Sending state 3.27500 V with 3 decimals of accuracy
[sensor:125]: 'ant-bms cell voltage 4': Sending state 3.27900 V with 3 decimals of accuracy
[sensor:125]: 'ant-bms cell voltage 5': Sending state 3.27900 V with 3 decimals of accuracy
[sensor:125]: 'ant-bms cell voltage 6': Sending state 3.28000 V with 3 decimals of accuracy
[sensor:125]: 'ant-bms cell voltage 7': Sending state 3.27900 V with 3 decimals of accuracy
[sensor:125]: 'ant-bms cell voltage 8': Sending state 3.26700 V with 3 decimals of accuracy
[sensor:125]: 'ant-bms current': Sending state 0.00000 A with 1 decimals of accuracy
[sensor:125]: 'ant-bms soc': Sending state 0.00000 % with 0 decimals of accuracy
[sensor:125]: 'ant-bms total battery capacity setting': Sending state 50.00000 Ah with 2 decimals of accuracy
[sensor:125]: 'ant-bms capacity remaining': Sending state 0.78125 Ah with 2 decimals of accuracy
[sensor:125]: 'ant-bms battery cycle capacity': Sending state 0.00300 Ah with 2 decimals of accuracy
[sensor:125]: 'ant-bms temperature 1': Sending state 14.00000 °C with 0 decimals of accuracy
[sensor:125]: 'ant-bms temperature 2': Sending state 13.00000 °C with 0 decimals of accuracy
[sensor:125]: 'ant-bms temperature 3': Sending state 12.00000 °C with 0 decimals of accuracy
[sensor:125]: 'ant-bms temperature 4': Sending state 10.00000 °C with 0 decimals of accuracy
[sensor:125]: 'ant-bms temperature 5': Sending state 65496.00000 °C with 0 decimals of accuracy
[sensor:125]: 'ant-bms temperature 6': Sending state 65496.00000 °C with 0 decimals of accuracy
[sensor:125]: 'ant-bms charge mosfet status code': Sending state 0.00000  with 0 decimals of accuracy
[text_sensor:067]: 'ant-bms charge mosfet status': Sending state 'Off'
[sensor:125]: 'ant-bms discharge mosfet status code': Sending state 0.00000  with 0 decimals of accuracy
[text_sensor:067]: 'ant-bms discharge mosfet status': Sending state 'Off'
[sensor:125]: 'ant-bms balancer status code': Sending state 0.00000  with 0 decimals of accuracy
[text_sensor:067]: 'ant-bms balancer status': Sending state 'Off'
[sensor:125]: 'ant-bms power': Sending state 999.90002 W with 1 decimals of accuracy
[sensor:125]: 'ant-bms min cell voltage': Sending state 3.28000 V with 3 decimals of accuracy
[sensor:125]: 'ant-bms max cell voltage': Sending state 3.26700 V with 3 decimals of accuracy
[sensor:125]: 'ant-bms average cell voltage': Sending state 3.27600 V with 2 decimals of accuracy
[sensor:125]: 'ant-bms battery strings': Sending state 8.00000  with 0 decimals of accuracy
```

syssi, 2022-02-23:
But it looks like my frame buffer to extract the frames doesn't work right now. I would be happy if you do some further tests. I will add some debug statements to the `ant_modbus` component to get a better understanding what's going on.

lucasimons, 2022-02-23:
> If I push your response data
>
> ``` [...]

i have 8 cells i can see only 8 voltages, i think that the power is wrong  and i don t have  temperature sensor 5 and 6

syssi, 2022-02-23:
I've pushed a fix. If we are lucky this custom component is working now. Please flash & try again. :-)

syssi, 2022-02-23:
I prepared a more verbose version. Please update your config yaml to

```
external_components:
  - source: github://syssi/esphome-ant-bms@debug-ant-modbus
    refresh: 0s
```

and flash & provide the log again.

lucasimons, 2022-02-23:
[logs_ant-bms_logs (4).txt](https://github.com/syssi/esphome-ant-bms/files/8127194/logs_ant-bms_logs.4.txt)

the sensors are all unknown

syssi, 2022-02-23:
I've updated the frame buffer. Please try again (the same feature branch):

```
external_components:
  - source: github://syssi/esphome-ant-bms@debug-ant-modbus
    refresh: 0s
```

lucasimons, 2022-02-23:
Now it works but he power is 999.9 w but it should be 0w

syssi, 2022-02-23:
Cool! I can debug the power sensor locally using your payload. Shouldn't be a big thing.

syssi, 2022-02-23:
FYI this custom component should be ESP8266 compatible. If you don't want to use a ESP32.

lucasimons, 2022-02-23:
i found a problem if i add these sensors in the .yaml file after the esp32 no longer connects

    temperature_5:
      name: "${name} temperature 5"
    temperature_6:
      name: "${name} temperature 6"

    cell_voltage_9:
      name: "${name} cell voltage 9"
    cell_voltage_10:
      name: "${name} cell voltage 10"
    cell_voltage_11:
      name: "${name} cell voltage 11"
    cell_voltage_12:
      name: "${name} cell voltage 12"
    cell_voltage_13:
      name: "${name} cell voltage 13"
    cell_voltage_14:
      name: "${name} cell voltage 14"
    cell_voltage_15:
      name: "${name} cell voltage 15"
    cell_voltage_16:
      name: "${name} cell voltage 16"
    cell_voltage_17:
      name: "${name} cell voltage 17"
    cell_voltage_18:
      name: "${name} cell voltage 18"
    cell_voltage_19:
      name: "${name} cell voltage 19"
    cell_voltage_20:
      name: "${name} cell voltage 20"
    cell_voltage_21:
      name: "${name} cell voltage 21"
    cell_voltage_22:
      name: "${name} cell voltage 22"
    cell_voltage_23:
      name: "${name} cell voltage 23"
    cell_voltage_24:
      name: "${name} cell voltage 24"
    cell_voltage_25:
      name: "${name} cell voltage 25"
    cell_voltage_26:
      name: "${name} cell voltage 26"
    cell_voltage_27:
      name: "${name} cell voltage 27"
    cell_voltage_28:
      name: "${name} cell voltage 28"
    cell_voltage_29:
      name: "${name} cell voltage 29"
    cell_voltage_30:
      name: "${name} cell voltage 30"
    cell_voltage_31:
      name: "${name} cell voltage 31"
    cell_voltage_32:
      name: "${name} cell voltage 32"
  
  charge_mosfet_status_code:
      name: "${name} charge mosfet status code"
    discharge_mosfet_status_code:
      name: "${name} discharge mosfet status code"
    balancer_status_code:
      name: "${name} balancer status code"

syssi, 2022-02-23:
You have to decrease the log level to free space / memory / stack. The maximum number of sensors is a bit limited. Enable the sensors step by step.

syssi, 2022-02-23:
Don't add useless sensors (voltages of cells you don't have) ;-)

syssi, 2022-02-23:
Try to avoid MQTT if you can and use the Home Assistant API (`api`).

syssi, 2022-02-23:
I've checked the power sensor. The calculation is fine. Your BMS reports:

```
0x00, 0x00, 0x27, 0x0F
```

This is `9999` in `dec` multiplied by '0.1f` is `999.9`. If you use the custom sensor of @SeByDocKy you should see the same value labeled as `ANTBMS_puiss`.

lucasimons, 2022-02-23:
[image]
I tried to change the charge and discharge switches via the app but when they are on the code is 0 but home assistant says off

syssi, 2022-02-23:
If you want to dig a bit deeper: You can enable the debug mode of the `uart` again and compare two responses (`charging off` vs. `charging on`). The `charging flag` should be hex value number `103`: 

https://github.com/klotztech/VBMS/wiki/Serial-protocol#live-data
https://github.com/syssi/esphome-ant-bms/blob/main/components/ant_bms/ant_bms.cpp#L131-L133

syssi, 2022-02-23:
I'm happy about every feedback so we can improve the documentation.

lucasimons, 2022-02-23:
> If you want to dig a bit deeper: You can enable the debug mode of the `uart` again and compare two responses (`charging off` vs. `charging on`). The `charging flag` should be hex value number [...]

I would like to try to see what my bms sends but the logger no longer works

[image]

syssi, 2022-02-23:
I've merged the feature branch into `main`. Please change the external component back to the `main` branch:

```
external_components:
  - source: github://syssi/esphome-ant-bms@main
    refresh: 0s
```

syssi, 2022-02-23:
Could you provide some details about your BMS (model description, ids, names whatever + ratings/specs)? I would like to extend the list of supported devices.

syssi, 2022-02-23:
> I tried to change the charge and discharge switches via the app but when they are on the code is 0 but home assistant says off

If you use the charge and discharge switches. Does the status codes change at all or is the status `0` always and doesn't change? If `0` means `On` and `1` means `Off` we can flip the values.

lucasimons, 2022-02-23:
> Could you provide some details about your BMS (model description, ids, names whatever + ratings/specs)? I would like to extend the list of supported devices.

this bms I tested with is a 16zmb-tb -7-16s-300a bought on aliexpress

in this log I started with discharge mosfet off then I turned it on and after it turned off it indicated manually turned of
[logs_ant-bms_upload.txt](https://github.com/syssi/esphome-ant-bms/files/8127661/logs_ant-bms_upload.txt)
f

SeByDocKy, 2022-02-23:
I think I have the same :)

lucasimons, 2022-02-23:
> If you use the charge and discharge switches. Does the status codes change at all or is the status `0` always and doesn't change? If `0` means `On` and `1` means `Off` we can flip the values.

i found the error i had to set the cell type via app now it works

lucasimons, 2022-02-23:
I would like to try why it shows the power 999.9w even if the current is 0A but currently I have made the temporary connections and I can not connect any load in order to check if the power value actually changes, however great job, 
I saw that you already have created this 

https://github.com/syssi/esphome-ant-bms/issues/5

to toggle mosfets later it would be perfect as bms jk integration

syssi, 2022-02-26:
Let's close this issue. Please create a new one if you have issues or miss something. I've answered you last question at the other issue.

closed 2022-02-26 by syssi

Stealthbomber76, 2022-09-09:
Hello. Im pretty new to antbms and looking for help! I got all working but i dont know how all happen but the auth. locked me out. Regardless how i wrote the 123456789abc or whatever it gave me error as it was wrong. Now i do believe the fw got messed up. Is there away to install fw again and how to do or where to download? Thanks

syssi, 2022-09-09:
@Stealthbomber76 Could you create a new issue? I will provide some support.

snuglinux, 2023-03-18:
I have a new model Ant BMS 10S to 24S, the connector is marked A, B, G, V, it looks like RS485/CAN. Can the ego also be connected directly to the ESP32 pins, or do you still need a TTL / RS485 board? https://www.aliexpress.com/item/32826820690.html

khvalera, 2023-03-18:
I have a new model Ant BMS 10S to 24S, the connector is marked A, B, G, V, it looks like RS485/CAN. Can the ego also be connected directly to the ESP32 pins, or do you still need a TTL / RS485 board? https://www.aliexpress.com/item/32826820690.html

syssi, 2023-03-19:
@dl4mea Could you help here? Which port do you use to read out your BMS via cable? It's a UART-TTL or RS485 signal?

@khvalera A high resolution photo of the USB-to-unknown dongle would be nice. I guess the IC / dongle is a USB-to-TTL converter.

syssi, 2023-03-19:
@khvalera Could you provide a photo of your BMS?

khvalera, 2023-03-20:
@syssi 
[image]
At the moment I can’t disassemble the BMS, since it’s in work, another device will arrive soon, I’ll try to unscrew the top cover and take a picture.

syssi, 2023-03-20:
Alright. Please keep me updated! I cannot help much with additional input because I don't own the specific device.

khvalera, 2023-03-20:
> Alright. Please keep me updated! I cannot help much without additional input because I don't own the specific device.

[image]
What part of the board needs to be photographed in more detail?

syssi, 2023-03-20:
Could you provide a photo of the PCB bottom too?

khvalera, 2023-03-20:
[image]
[image]

khvalera, 2023-03-20:
[image]

khvalera, 2023-03-20:
The board is covered with varnish, if the values \u200b\u200bof the chips are hard to see, I can try to remove it with a microcomputer, just tell me what exactly?
[image]
[image]

syssi, 2023-03-20:
74HC595AC

syssi, 2023-03-20:
Could you try to provide the label / or close ups of these chips:

[image]

[16 more replies]