syssi/esphome-ant-bms #5: Control the BMS

syssi, 2022-02-21:
```
-> 0xA5 0xA5 0xFA 0x00 0x00 0xFA  # Turn Off Charge MOSFET
-> 0xA5 0xA5 0xFA 0x00 0x010 xFB  # Turn On Charge MOSFET
-> 0xA5 0xA5 0xFE 0x00 0x00 0xFE  # Reboot BMS
-> 0xA5 0xA5 0xF9 0x00 0x00 0xF9  # Turn Off DisCharge MOSFET
-> 0xA5 0xA5 0xF9 0x00 0x01 0xFA  # Turn On DisCharge MOSFET
-> 0xA5 0xA5 0xFC 0x00 0x00 0xFC  # Toggles balancing
```

https://github.com/klotztech/VBMS/wiki/Serial-protocol#control-addresses
https://github.com/klotztech/VBMS/wiki/Serial-protocol#checksums

syssi, 2022-02-26:
@lucasimons It's complicated to implement it properly (command queueing/scheduling). We need some silence on the bus between two requests. At the moment there is only one request (asking for the status of the BMS) periodically. If we add an control element like a `switch` entity sending another request we have to make sure not to send both messages at the same time.

I can add the switch already (without scheduling) but you will see you have to toggle the switch multiple times before the setting gets accepting. The situation can be improved if you decrease the update interval of the status request but it's not the perfect solution at all.

lucasimons, 2022-02-26:
I did a test and when the balance is active the value becomes 4 so that's okay, but the switches don't do anything

[logs_ant-bms_logs.txt](https://github.com/syssi/esphome-ant-bms/files/8146278/logs_ant-bms_logs.txt)

syssi, 2022-02-26:
I've pushed a fix to `main`. Please flash and retry again. :-) 

https://github.com/syssi/esphome-ant-bms/commit/959f966cb6d8714692853a3d77c5319473e96bab

lucasimons, 2022-02-26:
I tried but nothing seems to have changed
[logs_ant-bms_logs (1).txt](https://github.com/syssi/esphome-ant-bms/files/8146346/logs_ant-bms_logs.1.txt)

syssi, 2022-02-26:
I've reverted the last change. Please flash again. Could you try to use the button multiple times in a row? Is the new value accepted sometimes?

lucasimons, 2022-02-26:
Ok now I try, but to use the button several times in a row what should I do?  I just have to toggle the switch several times

syssi, 2022-02-26:
Toggling the switch several times is fine!

syssi, 2022-02-26:
It's hard to understand. I will rephrase it: If you turn off the charging button f.e. and the button flips back to "on" after a moment please turn off charging again and again. At some point (if we have luck) the charging should stay off.

lucasimons, 2022-02-26:
I did some tests but without success, but I think something has changed from the [2021](https://www.youtube.com/watch?v=myGlZyOAtI4) version onwards because first the app made the switch change state without the need for a password, now instead to activate the charge or discharge mosfets and to activate the balancing you need to enter a password and after then you can intervene, instead in the old app you could do everything without the need for a password

[logs_ant-bms_logs (2).txt](https://github.com/syssi/esphome-ant-bms/files/8146407/logs_ant-bms_logs.2.txt)
[logs_ant-bms_upload.txt](https://github.com/syssi/esphome-ant-bms/files/8146408/logs_ant-bms_upload.txt)

syssi, 2022-02-26:
You you point me to the android APK of the 2021 version?

syssi, 2022-02-26:
I've checked the android app and found the root cause. It looks like there is no password required but every write command must be "applied" with a second apply message:

```
0xA5 0xA5 0xFA 0x00 0x00 0xFA # Turn charging on
0xA5 0xA5 0xFF 0x00 0x00 x0FF # Apply previous command
```

lucasimons, 2022-02-26:
I did some tests but it seems that the various buttons or switches do nothing

[logs_ant-bms_upload (1).txt](https://github.com/syssi/esphome-ant-bms/files/8147124/logs_ant-bms_upload.1.txt)

syssi, 2022-02-26:
I've added a delay between the two commands. I assume some separation is required here. Could you try this feature branch this time:

```
external_components:
  - source: github://syssi/esphome-ant-bms@test-timings
    refresh: 0s
```

If it works we try to reduce the delay. I hope 50ms is possible.

lucasimons, 2022-02-26:
Yes, I also have an old version that I use with my photovoltaic system for which they cannot solder the wires and I am waiting for the connector from China as soon as it arrives I do the tests with that, in the meantime I hope that other people can be added to expand the tests with other versions

syssi, 2022-02-27:
I've some new questions:

1. Are you able to control both BMS versions with the new app or do you have one app per BMS version at the moment?
2. Could you provide the firmware/software version of both BMS?

The background: It looks like the android app supports two protocols. The old one and the new password protected one. Furthermore it looks like the new BMS version supports two protocols (the old one = read-only) and the new password protected one (read/write). The windows application (`MaYiSystem.exe`) to monitor and control the BMS supported the new protocol only. I would like to identify the software version supporting the new protocol version.

lucasimons, 2022-02-27:
> Yes, I also have an old version that I use with my photovoltaic system for which they cannot solder the wires and I am waiting for the connector from China as soon as it arrives I do the tests with that, in the meantime I hope that other people can be added to expand the tests with other versions

@lumiror @HaraldRuppert @SeByDocKy

lucasimons, 2022-02-27:
this afternoon I will try to disconnect the raspberry from my old bms so I can try the new APP with the old bms, this is the version of the app I used with the old bms but it does not work with the new bms

[ANT_BMS_20180101.zip](https://github.com/syssi/esphome-ant-bms/files/8148402/ANT_BMS_20180101.zip)

syssi, 2022-02-27:
I've found this new version of the 2018er app (`com.lecheng.hello.ant_bms`): http://mysharegadget.com/734493564

This app (`antBMS_Android_English.apk`) contains some new commands to send a password (8 chars) using the old protocol.

SeByDocKy, 2022-02-27:
Yes I guess first, the password must be send as we do via the appz before to access controls....

syssi, 2022-02-27:
If you use the app do authorizes changes is it password 8 chars (`12345678`) or 12 chars (`123456789abc`) long?

lucasimons, 2022-02-27:
To turn the mosfet or balancing on or off, the password 12345678 is enough to change the parameters instead, the password 12345678abc is needed

syssi, 2022-02-27:
I've prepared a new feature branch called `add-auth`: 

```
external_components:
  - source: github://syssi/esphome-ant-bms@add-auth
    refresh: 0s
```

Your password must be added to the config yaml:

```
ant_bms:
  id: bms0
  ant_modbus_id: modbus0
  update_interval: 10s
  password: "12345678"
```

If there is no password available the authentication will be skipped. Please give it a try and provide some logs again. :-)

lucasimons, 2022-02-27:
nothing seems to have changed I also tried the balance

[logs_ant-bms_logs.txt](https://github.com/syssi/esphome-ant-bms/files/8148722/logs_ant-bms_logs.txt)

syssi, 2022-02-27:
I've pushed another small change. If it doesn't work this time I'm out of ideas.

SeByDocKy, 2022-02-27:
> I've pushed another small change. If it doesn't work this time I'm out of ideas.

Ok testing ....

SeByDocKy, 2022-02-27:
Not working :( :( :(

syssi, 2022-02-27:
Could somebody try this Android app: http://mysharegadget.com/734493564

Are you able to control the new BMS version with this app?

SeByDocKy, 2022-02-27:
Ok sorry .... I was talking about the JK BMS controls ....   I am in the antBMS thread here

syssi, 2022-02-27:
Alright. In this case we have a last chance. :-)

lucasimons, 2022-02-27:
> Could somebody try this Android app: http://mysharegadget.com/734493564
> 
> Are you able to control the new BMS version with this app?

This app version not works with the new bms i can see the value but I can't change the parameters or the mosfet switches 

In the new app for the new bms I find this software version 

[image]
[image]

syssi, 2022-02-27:
Your last test (using the `antBMS_Android_English.apk`) looks like the proof for:

1. The new BMS supports two protocols: The old one (read-only) and a new one (read/write)
2. The `antBMS_Android_English.apk` uses the old protocol exclusively (like the custom component) -> No control possible
3. The `My App 1116.zip` supports two protocols (the old one and the new one). To control the new one, the new protocol is used.

Summary: The control of the old BMS version should be possible now. To control the new version this protocol must be implemented: https://github.com/syssi/esphome-ant-bms/issues/20

syssi, 2022-02-27:
I've prepared a new attempt. Please use the feature `new-protocol-commands`:

```
external_components:
  - source: github://syssi/esphome-ant-bms@new-protocol-commands
    refresh: 0s
```

The password is hardcoded to `123456789abc` and the `charging` switch is the only supported entity at the moment. Could you flash the new version and press the charging switch a few times? Thanks for your patience!

lucasimons, 2022-02-27:
there is an invalid header when i switch the toggle
[logs_ant-bms_upload.txt](https://github.com/syssi/esphome-ant-bms/files/8149497/logs_ant-bms_upload.txt)

syssi, 2022-02-27:
This is fine. The response parser doesn't support the new response format. The charging mosfet should be enabled/disabled nevertheless. We are ignoring the answer of the BMS to the control command at the moment.

syssi, 2022-02-28:
Could you try the following:

1. Turn `charging` on via the Android App
2. Try to turn `charging` off once
3. Does the BMS accept the command and turns charging off?
4. If the switch flips back to `on` change it to `off` again + wait 15 seconds

Does the charging switch stay `off` at some point?

lucasimons, 2022-02-28:
Now it works but the charge switch turn on/off the discharge mosfet 
[logs_ant-bms_logs (1).txt](https://github.com/syssi/esphome-ant-bms/files/8152377/logs_ant-bms_logs.1.txt)

lucasimons, 2022-02-28:
[image][image]

lucasimons, 2022-02-28:
It works perfectly I can activate it via APP and it also changes on HA and vice versa ... Obviously only the Charge Mosfet Switch works

syssi, 2022-02-28:
Sure. I'm implementing a generic approach right now but I will take some time to get it done.

syssi, 2022-02-28:
Please update yaml configuration:

```
ant_bms:
  id: bms0
  ant_modbus_id: modbus0
  password: "${password}"
  update_interval: 10s
  supports_new_commands: true
```

The `supports_new_commands` parameter is important. There are some new switches:

```
switch:
  - platform: ant_bms
    charging:
      name: "${name} charging"
    discharging:
      name: "${name} discharging"
    balancer:
      name: "${name} balancer"
    bluetooth:
      name: "${name} bluetooth"
    buzzer:
      name: "${name} buzzer"
```

And sure to use the feature branch. The new commands aren't official supported/merged yet:

```
external_components:
  - source: github://syssi/esphome-ant-bms@new-protocol-commands
    refresh: 0s
```

lucasimons, 2022-02-28:
I tried to flash but there is this error
[image]

syssi, 2022-02-28:
I've removed the `buzzer` and `bluetooth` switch again. Please use:

```
switch:
  - platform: ant_bms
    charging:
      name: "${name} charging"
    discharging:
      name: "${name} discharging"
    balancer:
      name: "${name} balancer"
```

lucasimons, 2022-02-28:
Now this switches work great

> I've removed the `buzzer` and `bluetooth` switch again. Please use:
> 
> ```
> switch:
>   - platform: ant_bms
>     charging:
>       name: "${name} charging"
>     discharging:
>       name: "${name} discharging"
>     balancer:
>       name: "${name} balancer"
> ```

syssi, 2022-02-28:
I will update the response parser later the week to avoid the "invalid header" warnings.

syssi, 2022-03-04:
@lucasimons Could you re-flash your ESP (still) using this feature branch:

```
external_components:
  - source: github://syssi/esphome-ant-bms@new-protocol-commands
    refresh: 0s
```

and provide some logs? Best case: The "Invalid header" warning is gone and everything is still working. ;-)

lucasimons, 2022-03-04:
it works perfectly, I found a bug in the app 2021 in the interface the states of the mosfets are inverted
[logs_ant-bms_logs.txt](https://github.com/syssi/esphome-ant-bms/files/8185870/logs_ant-bms_logs.txt)

syssi, 2022-03-05:
I will close this issue because it's possible to control a BMS version 2021 now. I've created an follow-up issue to test and improve the pre 2021 BMS version.

commit bb634624 (2022-02-26): Add BMS controls (Closes: #5) (#14)
closed 2022-03-05 by syssi