syssi/esphome-ant-bms #15: Add missing sensors

syssi, 2022-02-26:
## Useful registers

```
  //  87    0x00 0x08 0x57 0x20: Uptime in seconds     546.592 s / 3600 = 151.83 h    s
  //  108   0x00 0x17: Number of pulses per week # meaning!?
  //  110   0x01: Relay switch
  //  115   0x0D: Cell with the highest voltage        Cell 13
  //  118   0x09: Cell with the lowest voltage         Cell 9
  //  132   0x00 0x00 0x00 0x00: Battery is in balance bitmask (Bit 1 = Cell 1, Bit 2 = Cell 2, ...)
  //  136   0x11 0x62: System log / overall status bitmask?
```

## Probably not important

```
  //  106   0x03 0xE8: Tire length                                                    mm
  //  124   0x00 0x00: Discharge MOSFET, voltage between D-S                          0.1 V
  //  126   0x00 0x73: Drive voltage (discharge MOSFET)                               0.1 V
  //  128   0x00 0x6F: Drive voltage (charge MOSFET)                                  0.1 V
  //  130   0x02 0xA7: When the detected current is 0, the initial value of the comparator
```

syssi, 2022-02-26:
@lucasimons Do you know the meaning of the `relay switch` register? Is the BMS equipped with a relay or something?

lucasimons, 2022-02-26:
in the bms there aren't relays so I don't know what it refers to to interrupt the current there are mosfets

lucasimons, 2022-03-04:
//  87    0x00 0x08 0x57 0x20: Uptime in seconds     546.592 s / 3600 = 151.83 h    s

This is the most useful sensor that has not been added

syssi, 2022-03-04:
Do you know the meaning of "Number of pulses per week"?

syssi, 2022-03-04:
@lucasimons Could you check something: Is the value the "current uptime" (since the last restart) or should we call the sensor "total runtime" / "operating hours"?

lucasimons, 2022-03-04:
I don't know what it could be ... It should be total Uptime

syssi, 2022-03-04:
Could you check your Android app and restart the BMS once + check again?

lucasimons, 2022-03-04:
it is a total uptime, i also checked via app to restart.....this is my 2020 bms integrated in HA via bluetooth  you can see that it has been running for 316 days
[image]

syssi, 2022-03-04:
@lucasimons The new sensor `total runtime` is available in `main` now. Please update your `external_components` source.

lucasimons, 2022-03-04:
Yeah, you can format the uptime sensor by days, hours, seconds? The balancer button what it should do?

syssi, 2022-03-04:
I don't want to expose the numeric value as formatted string. Please use a template sensor to apply some special format.

syssi, 2022-03-04:
Please adapt something like this: https://esphome.io/components/sensor/uptime.html#human-readable-sensor

```
  - platform: uptime
    name: Uptime Sensor
    id: uptime_sensor
    update_interval: 60s
    on_raw_value:
      then:
        - text_sensor.template.publish:
            id: uptime_human
            state: !lambda |-
              int seconds = round(id(uptime_sensor).raw_state);
              int days = seconds / (24 * 3600);
              seconds = seconds % (24 * 3600);
              int hours = seconds / 3600;
              seconds = seconds % 3600;
              int minutes = seconds /  60;
              seconds = seconds % 60;
              return (
                (days ? to_string(days) + "d " : "") +
                (hours ? to_string(hours) + "h " : "") +
                (minutes ? to_string(minutes) + "m " : "") +
                (to_string(seconds) + "s")
              ).c_str();
```

syssi, 2022-03-04:
Hmmm. I've changed my mind. I will implement a second sensor tomorrow to provide some formatted text.

lucasimons, 2022-03-04:
the second sensor is better than hour sensor, so  i can use this in HA
        bms_uptime:
            friendly_name: "Time BMS"    
            value_template: "{{ states('sensor.timebms') | int | timestamp_custom('%j d %H h %M m %S s', 0) }}"

syssi, 2022-03-04:
If you multiply the hours by 3600 you have seconds again. ;-)

syssi, 2022-03-05:
There is a second sensor called `total_runtime_formatted` now. The unit of the `total_runtime` sensor is `seconds` instead of `hours` now:

```
[text_sensor:067]: 'ant-bms total runtime formatted': Sending state '6d 7h '
[sensor:124]: 'ant-bms total runtime': Sending state 546592.00000 s with 0 decimals of accuracy
```

syssi, 2022-03-18:
I will close this issue because the added value of the other sensors is unknown.

closed 2022-03-18 by syssi