syssi/esphome-ant-bms #28: Control the BMS version pre 2021 (2020)

syssi, 2022-03-05:


syssi, 2022-03-14:
Please set `supports_new_commands` to `false` and provide a password:

```
ant_bms:
  id: bms0
  ant_modbus_id: modbus0
  password: "12345678"
  update_interval: 10s
  supports_new_commands: false
```

And don't add this switch to the config because it's not supported by the BMS version:

```
switch:
    balancer:
      name: "${name} balancer"
```

Instead you have to use the trigger button:

```
button:
    balancer:
      name: "${name} balancer"
```

@lucasimons let's proceed here.

lucasimons, 2022-03-16:
I'm sorry but I use this battery for home it also power supply  the modem, router so I can't test the discharge switch, but I tested the charge switch and it works the only thing that is wrong, it is the power but the current is right for example 1A 27V = 27W instead marks 2.7W so there is a factor of 10 of erro
[logs_ant-bms-2020_logs.txt](https://github.com/syssi/esphome-ant-bms/files/8260454/logs_ant-bms-2020_logs.txt)
r

syssi, 2022-03-16:
Thanks for your support. The log looks fine. Just some minor things/dotos:

1. I would like to fix:

```
[09:58:09][D][switch:017]: 'ant-bms-20 charging' Turning OFF.
[09:58:09][D][ant_bms:231]: Authenticate write command
[09:58:09][W][ant_modbus:070]: Invalid header.
[09:58:09][W][ant_modbus:070]: Invalid header.
[09:58:09][W][ant_modbus:070]: Invalid header.
[09:58:09][W][ant_modbus:070]: Invalid header.
[09:58:10][W][ant_modbus:070]: Invalid header.
```

The BMS response of the control instruction isn't handled properly yet. Could you enable the debug output of the `uart` component and provide an example response / or a full log turning charging on/off again. 

2. I will fix the factor of the power measurement. Could you check the power sensor of your new BMS version? I would like to know we need a special case here or it's wrong for both versions.

3. Could you provide the model name / type and specs of your old BMS version? I would like to add it to the list of supported devices.

lucasimons, 2022-03-16:
This is my 2020 bms there isn t any model or label... 
[image]

syssi, 2022-03-16:
The power sensor is fixed!

lucasimons, 2022-03-17:
There always seems to be a problem with potency perhaps because it is negative? it should be around -250w

[logs_ant-bms-2020_logs.txt](https://github.com/syssi/esphome-ant-bms/files/8283047/logs_ant-bms-2020_logs.txt)

syssi, 2022-03-17:
```
111 FF
112 FF
113 FE
114 E3

# -214748352.00000 A
```

lucasimons, 2022-03-17:
Now i m using this sensor with a rpi script, and it works fine

#Power
 try:
    power = (Antw33.encode('hex') [(111*2):(114*2+2)])
    if int(power,16)>2147483648:
      power=(-(2*2147483648)+int(power,16))
      print power
      client.publish("battery/power", str(power))
    else:
      power=int(power,16)
      print power
      client.publish("battery/power", str(power))
 except:
    pass 
 
 #BMS current
 try:
    amps = (Antw33.encode('hex') [(70*2):(73*2+2)])
    if int(amps,16)>2147483648:
      amps=(-(2*2147483648)+int(amps,16))*0.1
      print amps
      client.publish("battery/amps", amps)
    else:
      amps = int(amps,16)*0.1
      print amps
      client.publish("battery/amps", amps)
 except:
    pass
#MOSFET Statu

syssi, 2022-03-17:
Could you try this feature branch:

```
external_components:
  - source: github://syssi/esphome-ant-bms@fix-negative-power
    refresh: 0s
```

This should fix the current and power sensor.

syssi, 2022-03-17:
I've verified my change locally. Both sensors are fine now.

lucasimons, 2022-03-17:
I installed it and for now the power is right tomorrow with the sun when it is negative I will update you

syssi, 2022-03-18:
Did you change your wiring or could you double check your connections? The received FF:FF:FF:... bytes aren't valid data. It looks like noise, long wires or bad connection. You can control the update interval at the yaml (`update_interval` of the `ant-bms` component section). Please make sure your connection is stable first. At the moment increasing the interval won't help.

syssi, 2022-03-18:
This is the setting I'm talking about: https://github.com/syssi/esphome-ant-bms/blob/2471dcc1b8a2942052ccd30962a893a2aaba99f3/esp32-example.yaml#L47

lucasimons, 2022-03-18:
In reality I have not changed anything, everything has been active for about 30h but I really want to use an esp32 instead of the d1 mini, if I wanted to use this component for the bms, then the component for the pip 2424 inverter and also the component for the mppt victron solar charge controller , all on the same esp32 to which gpio is it better to configure the various uart? GPIO16/17 GPIO38/39 for the last serial connection I can choose any pair of gpio?

lucasimons, 2022-03-19:
I am doing some tests but it seems when I add the sensors of this component the esp 32 does not start

```
uart:

- id: uart_bus
  tx_pin: GPIO22 # not connected !! 
  rx_pin: GPIO34 # d7 use olny this pin and GND to MPPT chrager
  baud_rate: 19200 
  stop_bits: 1
  data_bits: 8
  parity: NONE
  rx_buffer_size: 256
  
  
- id: uart_pzem
  rx_pin: RX
  tx_pin: TX
  baud_rate: 9600

- id: uart0
  baud_rate: 19200
  rx_buffer_size: 384
  tx_pin: GPIO17
  rx_pin: GPIO16

ant_modbus:
  id: modbus0
  uart_id: uart0

ant_bms:
  id: bms0
  ant_modbus_id: modbus0
  password: "12345678"
  update_interval: 10s
  supports_new_commands: false
  

modbus:
- uart_id: uart_pzem
  id: modbus_1 # needed only for multiple buses

victron:
  id: victron0
  uart_id: uart_bus  

sensor:  

  - platform: victron
    victron_id: victron0
    panel_voltage:
      name: "Tensione Fotovoltaico 2" 
      id: pv
      filters:
        - sliding_window_moving_average:
            window_size: 10
            send_every: 10
    battery_voltage:
      name: "Batteria Tensione"  
      id: bv
      filters:
        - sliding_window_moving_average:
            window_size: 30
            send_every: 30    
    battery_current:
      name: "Batteria Corrente" 
      id: bc
      filters:
        - sliding_window_moving_average:
            window_size: 30
            send_every: 30        
    load_current:
      name: "Corrente carichi DC" 
      id: load_current
      filters:
        - sliding_window_moving_average:
            window_size: 10
            send_every: 10     
    max_power_yesterday:  
      name: "Massima potenza Ieri" 
      id: max_power_yesterday
      filters:
        - sliding_window_moving_average:
            window_size: 300
            send_every: 300       
    max_power_today:  
      name: "Massima potenza Oggi" 
      id: max_power_today
      accuracy_decimals: 2
      filters:
        - sliding_window_moving_average:
            window_size: 300
            send_every: 300        
    yield_total:
      name: "Energia Totale Victron"   
      id: yield_total
      filters:
        # Multiplication factor from W to kW is 0.001
        - multiply: 0.001
        - sliding_window_moving_average:
            window_size: 300
            send_every: 300 
      unit_of_measurement: kWh
      accuracy_decimals: 2
    yield_yesterday:
      name: "Energia FV2 Ieri"
      filters:
        # Multiplication factor from W to kW is 0.001
        - multiply: 0.001
        - sliding_window_moving_average:
            window_size: 300
            send_every: 300 
      unit_of_measurement: kWh
      accuracy_decimals: 2
    yield_today:
      name: "Energia FV2 Oggi"  
      id: yt
      filters:
        # Multiplication factor from W to kW is 0.001
        - multiply: 0.001
        - sliding_window_moving_average:
            window_size: 300
            send_every: 300         
      unit_of_measurement: kWh
      accuracy_decimals: 2
    panel_power:
      name: "Potenza Fotovoltaico 2"  
      id: panel_power
      filters:
        - sliding_window_moving_average:
            window_size: 10
            send_every: 10    
    day_number:
      name: "Day sequence number"  
      id: day_number
      icon: "mdi:calendar-today"
    charging_mode_id:
      name: "Charger mode id"
    error_code:
      name: "Error code"
    tracking_mode_id:
      name: "Tracker mode id"
  - platform: wifi_signal
    name: "Fotovoltaico WiFi Signal Sensor"
    id: rssi_sensor
    update_interval: 5s 
    
    
  - platform: pzemac
    modbus_id: modbus_1
    current:
      name: "Inverter Corrente"
    voltage:
      name: "Inverter Tensione"
    energy:
      name: "Inverter Energia"
    power:
      name: "Inverter Potenza"
    frequency:
      name: "Inverter Frequenza"
    power_factor:
      name: "Inverter Power Factor"
    update_interval: 10s

  - platform: ant_bms
    battery_strings:
      name: "bms1 battery strings"
    current:
      name: "bms1 current"
    soc:
      name: "bms1 soc"

    capacity_remaining:
      name: "bms1 capacity remaining"
    battery_cycle_capacity:
      name: "bms1 battery cycle capacity"
    total_voltage:
      name: "bms1 total voltage"
    total_runtime:
      name: "bms1 total runtime"
    average_cell_voltage:
      name: "bms1 average cell voltage"
    power:
      name: "bms1 power"
    min_cell_voltage:
      name: "bms1 min cell voltage"
    max_cell_voltage:
      name: "bms1 max cell voltage"
    min_voltage_cell:
      name: "bms1 min voltage cell"
    max_voltage_cell:
      name: "bms1 max voltage cell"
    temperature_1:
      name: "bms1 temperature 1"
    temperature_2:
      name: "bms1 temperature 2"
    temperature_3:
      name: "bms1 temperature 3"
    temperature_4:
      name: "bms1 temperature 4"

    cell_voltage_1:
      name: "bms1 cell voltage 1"
    cell_voltage_2:
      name: "bms1 cell voltage 2"
    cell_voltage_3:
      name: "bms1 cell voltage 3"
    cell_voltage_4:
      name: "bms1 cell voltage 4"
    cell_voltage_5:
      name: "bms1 cell voltage 5"
    cell_voltage_6:
      name: "bms1 cell voltage 6"
    cell_voltage_7:
      name: "bms1 cell voltage 7"
    cell_voltage_8:
      name: "bms1 cell voltage 8"

    charge_mosfet_status_code:
      name: "bms1 charge mosfet status code"
    discharge_mosfet_status_code:
      name: "bms1 discharge mosfet status code"
    balancer_status_code:
      name: "bms1 balancer status code"

```

syssi, 2022-03-19:
Please begin with 3 sensors per connected device (9 sensors in total) and try again. Does it work? Are you able to talk to your devices and are you able to update the ESP via OTA? If everything is fine we can move on.

syssi, 2022-03-19:
The max number of sensors is a bit limited (per default) if you use a ESP32. @SeByDocKy is using this fix to allow additional sensors (by increasing the stack size):

```
esphome:
  name: pvbrain
  platformio_options:
    build_flags: 
      - -DCONFIG_ARDUINO_LOOP_STACK_SIZE=32768 #16384 #8192 #16384 # #  
    platform_packages:
      - framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#2.0.2
    platform:
      - https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream  
```

Please give it a try (later!). First please make sure your setup is stable and working with just a few sensors.

SeByDocKy, 2022-03-19:
> The max number of sensors is a bit limited (per default) if you use a ESP32. @SeByDocKy is using this fix to allow additional sensors (by increasing the stack size):
>
> ```
> esphome:
>   name: pvbrain [...]

In order to compile it correctlyn you need a linux/windws box. These options are not (yet) working with Hassio

lucasimons, 2022-03-20:
I tried to add the sensors to 5/6 at a time and it seems to work, I noticed that the max and min cell voltage are reversed, then I wanted to know if I should use this new branch or can I also install the main one?

[image]

syssi, 2022-03-20:
Please use the main branch. I've fixed this issue right now!

lucasimons, 2022-03-20:
ok now I install the main branch, but I noticed that the problem is in the text sensors because I can't add both the platform

```
text_sensor:
  - platform: ant_bms
    charge_mosfet_status:
      name: "bms1 charge mosfet status"
    discharge_mosfet_status:
      name: "bms1 discharge mosfet status"
    balancer_status:
      name: "bms1 balancer status"
    total_runtime_formatted:
      name: "bms1 total runtime formatted"
      
 
 -  platform: victron
    victron_id: victron0
    charging_mode:
      name: "Charging Mode"
    error:
      name: "Error"
    tracking_mode:
      name: "Tracking Mode"

```

[image]

lucasimons, 2022-03-24:
These days I was doing tests to remove the raspberry and use an esp8266 or esp32 to read the bms data but I noticed many disconnections, the uptime when it reaches 1000 seconds the esp restarts causing a reset of the sensors, I did various tests then, 1 test d1mini with external Ant component installed and it had disconnections I tried to change the board with the one on which I had done the inverter tests but when I insert this external component it does not work, I also did a test with this [victron](https://github.com/KinDR007/VictronMPPT-ESPHOME) esp32 dev board

[image]

syssi, 2022-03-24:
Please check the WiFi signal strength and move an access point next to the ESP just to rule out WiFi issues.

lucasimons, 2022-03-24:
It is not a signal strength problem because I am at a distance of 1m from the router 7530 ax, then I did the test on two d1 mini boards but also on nodemcu v2 and esp32 dev and in all cases it becomes unstable if you have one of these board I believe that the test is replicable even if you do not have the inverter and the bms because the board becomes unstable it disconnects from HA and no longer reads anything

syssi, 2022-03-24:
Is my understanding correct: The setup is always unstable as soon the external component "esphome-ant-bms" is used and it doesn't depend on the used ESP model (ESP32 vs. ESP8266)? An accurate and honest answer is important to narrow down the problem. If the conclusion is: This custom component doesn't work stable please create a new issue and provide your configuration yaml.

syssi, 2022-04-03:
This issue is solved. Feel free to create new issues if something is missing or doesn't work as expected!

closed 2022-04-03 by syssi