syssi/esphome-ant-bms #33: Stability issue

lucasimons, 2022-03-24:
this is my config

```
esphome:
  name: sottoscala
  platform: ESP8266
  board: nodemcuv2
  
  
external_components:
  - source: github://syssi/esphome-ant-bms@main
    refresh: 0s
  - source: github://andreashergert1984/esphome@test_newcrc
    refresh: 0s
    components: [pipsolar]
# Enable logging
logger:
  baud_rate: 0

# Enable Home Assistant API
api:

[logs_sottoscala_logs.txt](https://github.com/syssi/esphome-ant-bms/files/8339509/logs_sottoscala_logs.txt)

captive_portal:

uart:
- id: uart0
  baud_rate: 19200
  rx_buffer_size: 384
  tx_pin: GPIO1
  rx_pin: GPIO3
  debug:
    direction: BOTH

ant_modbus:
  id: modbus0
  uart_id: uart0

ant_bms:
  id: bms0
  ant_modbus_id: modbus0
  password: "12345678"
  update_interval: 10s
  supports_new_commands: false

sensor:
 
      
  - platform: ant_bms

    current:
      name: "bms1 current"
      filters:
        - sliding_window_moving_average:
            window_size: 10
            send_every: 10       
    soc:
      name: "bms1 soc"
      filters:
        - sliding_window_moving_average:
            window_size: 10
            send_every: 10   
    capacity_remaining:
      name: "bms1 capacity remaining"
      filters:
        - sliding_window_moving_average:
            window_size: 10
            send_every: 10       
    battery_cycle_capacity:
      name: "bms1 battery cycle capacity"
      filters:
        - sliding_window_moving_average:
            window_size: 10
            send_every: 10       
    total_voltage:
      name: "bms1 total voltage"
      filters:
        - sliding_window_moving_average:
            window_size: 10
            send_every: 10       
    total_runtime:
      name: "bms1 total runtime"
    average_cell_voltage:
      name: "bms1 average cell voltage"
      filters:
       - sliding_window_moving_average:
           window_size: 10
           send_every: 10   
    power:
      name: "bms1 power"
     
    min_cell_voltage:
      name: "bms1 min cell voltage"
      filters:
        - sliding_window_moving_average:
            window_size: 10
            send_every: 10       
    max_cell_voltage:
      name: "bms1 max cell voltage"
      filters:
        - sliding_window_moving_average:
            window_size: 10
            send_every: 10       
    min_voltage_cell:
      name: "bms1 min voltage cell"
      filters:
        - sliding_window_moving_average:
            window_size: 10
            send_every: 10       
    max_voltage_cell:
      name: "bms1 max voltage cell"
      filters:
        - sliding_window_moving_average:
            window_size: 10
            send_every: 10       
    temperature_1:
      name: "bms1 temperature 1"
      filters:
        - sliding_window_moving_average:
            window_size: 10
            send_every: 10       
    temperature_2:
      name: "bms1 temperature 2"
      filters:
        - sliding_window_moving_average:
            window_size: 10
            send_every: 10       
    temperature_3:
      name: "bms1 temperature 3"
      filters:
        - sliding_window_moving_average:
            window_size: 10
            send_every: 10       
    temperature_4:
      name: "bms1 temperature 4"
      filters:
        - sliding_window_moving_average:
            window_size: 10
            send_every: 10 
    cell_voltage_1:
      name: "bms1 cell voltage 1"
      filters:
        - sliding_window_moving_average:
            window_size: 10
            send_every: 10       
    cell_voltage_2:
      name: "bms1 cell voltage 2"
      filters:
        - sliding_window_moving_average:
            window_size: 10
            send_every: 10       
    cell_voltage_3:
      name: "bms1 cell voltage 3" 
      filters:
        - sliding_window_moving_average:
            window_size: 10
            send_every: 10       
    cell_voltage_4:
      name: "bms1 cell voltage 4"
      filters:
        - sliding_window_moving_average:
            window_size: 10
            send_every: 10       
    cell_voltage_5:
      name: "bms1 cell voltage 5"
      filters:
        - sliding_window_moving_average:
            window_size: 10
            send_every: 10       
    cell_voltage_6:
      name: "bms1 cell voltage 6"
      filters:
        - sliding_window_moving_average:
            window_size: 10
            send_every: 10       
    cell_voltage_7:
      name: "bms1 cell voltage 7"
      filters:
        - sliding_window_moving_average:
            window_size: 10
            send_every: 10       
    cell_voltage_8:
      name: "bms1 cell voltage 8"
      filters:
        - sliding_window_moving_average:
            window_size: 10
            send_every: 10 
    charge_mosfet_status_code:
      name: "bms1 charge mosfet status code"
    discharge_mosfet_status_code:
      name: "bms1 discharge mosfet status code"
    balancer_status_code:
      name: "bms1 balancer status code"
      
  - platform: uptime
    name: "Ant  Uptime"
    id: uptime_s
    update_interval: 5s          
      
      
text_sensor:

  - platform: ant_bms
    charge_mosfet_status:
      name: "bms1 charge mosfet status"
    discharge_mosfet_status:
      name: "bms1 discharge mosfet status"
    balancer_status:
      name: "bms1 balancer status"
    total_runtime_formatted:
      name: "bms1 total runtime formatted"  

button:

      
      
  - platform: ant_bms

    balancer:
      name: "bms1 balancer"  
```

syssi, 2022-03-24:
Alright! I will prepare a test setup and implement some dummy BMS to reproduce the issue. Strange thing: The ESP doesn't crash. Just the WiFi connection drops evey now and then. I assume something is blocking the main loop. More strangeness: A ESP32 has two cores. The application runs at the second core. The WiFi stack at the first core. Blocking WiFi by bad application code shouldn't be possible here.

syssi, 2022-03-24:
As first step please remove this lines from your config:

```
  debug:
    direction: BOTH
```

Logging (the UART traffic) is super heavy and should be avoided if you don't wont to debug the traffic at all.

lucasimons, 2022-03-24:
I removed the string and I'm monitoring the uptime because yesterday on a d1 mini every 20 min it came off today I'm doing tests with a nodemcu v2

[image]

syssi, 2022-03-24:
Could you explain the graph? I cannot follow. This doesn't look like an uptime chart. ;-)

lucasimons, 2022-03-24:
Did you do any tests too? what do you recommend to try the bms 2021 to leave it turn on and see if that gives problems?

[image]

syssi, 2022-03-24:
This is my test setup:

```
substitutions:
  name: ant-bms-tx
  tx_pin: GPIO1
  rx_pin: GPIO3

esphome:
  name: ${name}
  platform: ESP8266
  board: d1_mini

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

api:
ota:
logger:
  baud_rate: 0
  level: DEBUG

uart:
  baud_rate: 19200
  tx_pin: ${tx_pin}
  rx_pin: ${rx_pin}
  debug:
    direction: BOTH
    dummy_receiver: true

interval:
  - interval: 1s
    then:
      - uart.write: [0xAA, 0x55, 0xAA, 0xFF, 0x01, 0x0B, 0x0D, 0x15, 0x0D, 0x14, 0x0D, 0x12, 0x0D, 0x11, 0x0D, 0x13, 0x0D, 0x12, 0x0D, 0x13, 0x0D, 0x13, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0xFF, 0xFF, 0xFF, 0x45, 0x25, 0x00, 0x00, 0x00, 0x00, 0x02, 0x9B, 0xAF, 0x0F, 0x01, 0x2F, 0x01, 0xF1, 0x01, 0xBA, 0x9E, 0xAB, 0x00, 0x11, 0x00, 0x11, 0x00, 0x0E, 0x00, 0x0F, 0x00, 0x00, 0x00, 0x00, 0x01, 0x01, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0xFF, 0xFF, 0xFE, 0x15, 0x01, 0x0D, 0x15, 0x04, 0x0D, 0x11, 0x0D, 0x12, 0x08, 0x00, 0x00, 0x00, 0x7E, 0x00, 0x7A, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x40, 0x05, 0x0D, 0xEE]
```

And you configuration yaml flashed to another ESP8266:

```
esphome:
  name: ant-bms-rx
  platform: ESP8266
  board: d1_mini

external_components:
  - source: github://syssi/esphome-ant-bms@main
    refresh: 0s
  - source: github://andreashergert1984/esphome@test_newcrc
    refresh: 0s
    components: [pipsolar]

logger:
  baud_rate: 0

api:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

captive_portal:

uart:
  - id: uart0
    baud_rate: 19200
    rx_buffer_size: 384
    tx_pin: GPIO1
    rx_pin: GPIO3
    debug:
[... 155 lines cut ...]
    balancer_status_code:
      name: "bms1 balancer status code"
  - platform: uptime
    name: "Ant  Uptime"
    id: uptime_s
    update_interval: 5s

text_sensor:
  - platform: ant_bms
    charge_mosfet_status:
      name: "bms1 charge mosfet status"
    discharge_mosfet_status:
      name: "bms1 discharge mosfet status"
    balancer_status:
      name: "bms1 balancer status"
    total_runtime_formatted:
      name: "bms1 total runtime formatted"

button:
  - platform: ant_bms
    balancer:
      name: "bms1 balancer"
```
[image]
[image]

syssi, 2022-03-24:
I had to care about my day job first. ;-) Good news for you: I can reproduce the issue:

[image]

lucasimons, 2022-03-24:
I also do this test but I had inserted the filters because I had the disconnections

syssi, 2022-03-24:
I've added an additional sensor to monitor the `api` connection:

```
binary_sensor:
  - platform: status
    name: "bms1 status"
```

And I've added some different BMS responses to the dummy BMS: https://github.com/syssi/esphome-ant-bms/blob/main/esp8266-fake-bms.yaml

I will report tomorrow the number of disconnects / issues. For now it looks stable.

syssi, 2022-03-25:
My test setup is stable. It does not restart and can keep the WLAN connection to the HA core.

lucasimons, 2022-03-25:
Now it seems to be stable, have you changed anything?   I just remove the filters... However, during the day I try to do the test with the esp32 and use 3 uart and see how it behaves

syssi, 2022-03-25:
No changes except removing the filters from your configuration.

https://github.com/syssi/esphome-ant-bms/commits/main

syssi, 2022-03-25:
Please feel free to create new issues. We should try to avoid off-topic discussions to keep things/threads clean and simple. :-)

closed 2022-03-25 by syssi

lucasimons, 2022-03-29:
every now and then I have some reading problems as I use the soc% for some automations and turn off the inverter for example, I would like to know if there is a way to set the last value and prevent it from going to 0%

[image]

syssi, 2022-03-29:
This is a bit strange. Could you enable the debug output of the UART component again and record some logs? In need a log containing 1-2 zero readings/failures.

syssi, 2022-03-29:
As workaround you could use the `filter_out` filter and throw zeros away. You automation should trigger on SoC 1% instead because there are no zero readings anymore:

```
filters:
  - filter_out: 0.0
```

lucasimons, 2022-03-29:
i think that it is a problem of connection or api because i see the same value to 0 of the voltages

[image]

syssi, 2022-03-29:
If we are able to identify f.e. 5-6 sensors (all zero at the same response) we could reject/skip the response to rule out these readings.