syssi/esphome-ant-bms #34: old ANT-BMS via Bluetooth

rdaguerre, 2022-03-29:
Hello, is there a possibility to adapt it to use in the old ANT-BMS, Bluetooth version? Thank you

syssi, 2022-03-29:
Could you describe your BMS model? Your version doesn't have a serial interface and can be controlled via Bluetooth only? Could you provide the APK of the Android app you are using to monitor/control the BMS?

lucasimons, 2022-03-29:
all version of ant bms have the serial interface and bluetooth

rdaguerre, 2022-03-29:
The App is My-BMS. Where can i use the serial interface? i try in Display port but not work
thanks.

[image]
[image]
S

lucasimons, 2022-03-29:
The serial connection is  the one for the display

rdaguerre, 2022-03-29:
i use this adapter and not work the comunication 
[image]

syssi, 2022-03-30:
This implementation must be flashed to a ESP32 oder ESP8266. The USB-to-TTL converter is only useful if you want to connect the BMS to a computer. Could you provide a photo of your ESP connected to the BMS? So we can check your wiring is correct.

rdaguerre, 2022-03-31:
Hi, thanks for the help. I managed to connect it directly from the Heltec esp32 to the old ANT-BMS and it works. The problem I'm having is that I can't get the bms to turn on. What should the ignition combination be? As for the wiring, it is the one that is explained. Thank you

syssi, 2022-04-01:
If the esphome node loses the WiFi connection / the API connection to HA gets disconnected the entities will be shown as unavailable. As soon the connection to the ESP is back the sensors should be available again.

Could you explain the "how to turn the BMS on" question" a bit more in detail? If the BMS is powered off you want be able to start the BMS remotely or are you asking about the way to turn on the BMS in general (if it's powered off)?

syssi, 2022-04-02:
@lucasimons Could you provide some details how to start the BMS? This is my understanding:

If you connect 3.3V and GND to the display port of the BMS the device should boot. If you add RX + TX you are able to read some data. If you disconnect 3.3V from the display port again the BMS don't want to communicate via RX/TX anymore. Can you confirm this?

@rdaguerre If you want to connect the display port "+" to some `GPIO` to remote control the BMS startup please make sure the "+" connector don't draw much current. If you see any current here you should use a transistor for switching via the GPIO. You are the first who makes this attempt!

lucasimons, 2022-04-02:
> @lucasimons Could you provide some details how to start the BMS? This is my understanding:
> 
> If you connect `3.3V` and `GND` to the display port of the BMS the device should boot. If you connect `RX`, `TX` you are able to query data. If you disconnect `3.3V` from the display port again the BMS don't want to communicate via RX/TX anymore. Can you confirm this?
> 
> @rdaguerre If you want to connect the display port `+` to some `GPIO` to remote control the BMS startup please make sure the `+` connector don't draw much current. If you see any current here you should use a transistor for switching via the GPIO. You are the first who makes this attempt!

So if the bms is already on you can see the bluetooth, otherwise you have to feed it to 3.3 as in the figure but in reality to read the data you have to feed the two pins on the left and connect the two on the right to tx and rx

[image]

closed 2022-04-09 by syssi