syssi/esphome-ant-bms #35: Reject dubious BMS responses

syssi, 2022-03-29:
Sporadically I have some reading problems as I use the soc% for some automations and turn off the inverter for example, I would like to know if there is a way to set the last value and prevent it from going to 0%

[image]

_Originally posted by @lucasimons in https://github.com/syssi/esphome-ant-bms/issues/33#issuecomment-1081761121_

syssi, 2022-03-29:
@lucasimons lets tackle this issue here.

syssi, 2022-03-29:
@lucasimons Could you provide screenshots of your total runtime and total energy sensor? Are there also zero readings visible?

lucasimons, 2022-03-29:
Unfortunately for these sensors I have not activated the recorder, however I have investigated and in reality it is not a problem with this custom component because it also happened to the values ​​of the inverter, so one could hypothesize it was the d1 mini where I connected both the inverter and the bms but in reality I think it is a problem given by the HA sensor templates

lucasimons, 2022-03-29:
I am using these sensor templates so as not to have to change all the lovelace entities, the mqtt sensors were of the old integration through rpi, in this way I changed from p3 to p2 and I made everything go with the rpi or through esp, now it seems that the custom components are stable and I could permanently change all the entities also from lovelace and from the automations but it would not be bad to have a second way if the custom componet no longer works, do you have something to recommend?
```
        powerbattery:
            unit_of_measurement: 'W'
            value_template: >-
              {% set p3 = states('sensor.powerbatterymqtt') | float(default=0) %}
              {% set p2 = states('sensor.bms1_power') | float(default=0) %}

                {% if is_state('sensor.masterstatus', 'unknown') %}
                  {{ (p2)|round(3, default=0) }}
                {% else %}
                  {{ (p3)|round(3, default=0) }}
                {% endif %}              
              

```

p3 old integration rpi & mqtt 
p2 new integration esp8266

lucasimons, 2022-03-29:
this is a switch connected near the inverter and the bms but on an esp32 where I also have the victron custom component and this has had 4 disconnections the two orange ones are at the same time where the SOC went to 0%, I think at this point it is a sensor template problem, however, I also had a disconnection of the swich
[image]

syssi, 2022-03-30:
My knowledge about sensor templates isn't pretty good. I will try to outline some ideas/edge-cases:

1. If you restart Home Assistant the sensor `sensor.masterstatus` isn't available instantly. You template will use the "else" path in this case and some `sensor.powerbatterymqtt` values will be published. If `sensor.powerbatterymqtt` isn't available too the `round("unknown", default=0)` call will fail and the sensor template will publish the fallback/default `0`.

2. If `p2` or `p3` doesn't provide a float the `round("unknown", default=0)` call will fail and `0` is published. I asume the sensor state is `unknown` or `unavailable` if the WiFi/API connection fails for a second.

3. Casting `p2` and `p3` to float and if this doesn't work return `0` is also a bad idea if you don't wont the `0` reading at your graphs. You are eliminating `none`/`unknown`/`unavailable` states here.

lucasimons, 2022-03-30:
So if I use this configuration I shouldn't have any problems, but it is strange that the same thing happens also to the switches which are not inserted in a sensor template

```
        soc:
            unit_of_measurement: '%'
            value_template: >-
              {% set p3 = states('sensor.socmqtt') | float(default=0) %}
              {% set p2 = states('sensor.bms1_soc') %}
                {% if is_state('sensor.masterstatus', 'unknown') %}
                  {{ (p2) }}
                {% else %}
                  {{ (p3)|round(0, default=0) }}
                {% endif %} 

```

syssi, 2022-03-30:
The case `sensor.masterstatus` is `unknown` and `p3` not properly castable will still result in `0` readings.

closed 2022-03-30 by syssi