syssi/esphome-ant-bms #36: Broken wifi connection

lumiror, 2022-03-31:
Hi, I don't know what I'm doing wrong, but the installation will end with a broken wifi connection. Can you advise on what the problem may be? Or how should I install it properly? Thank you
[image]
[antbms-new.txt](https://github.com/syssi/esphome-ant-bms/files/8392112/antbms-new.txt)

syssi, 2022-03-31:
Another idea: Please reduce the number of sensors/entities (to a minimum) just for testing. Is the setup stable afterwards?

syssi, 2022-03-31:
Alright. Please increase the number slowly. I will provide tomorrow a special config setting to increase the stack size of the ESP32 to allow a lot of sensors. At the moment the total number of sensors on the ESP32 platform is a bit limited. I will provide a link to the ESPHome issue too.

syssi, 2022-03-31:
https://github.com/esphome/issues/issues/855

syssi, 2022-03-31:
If everything is working a expected I would be happy about the model name or a photo of your ANT-BMS model to extend supported device section of the documentation.

lumiror, 2022-03-31:
I'll try tomorrow, thank you for now

syssi, 2022-04-01:
This is some additional configuration to increase the stack size of an **ESP32** to have room for a lot of sensor entities:

https://github.com/esphome/issues/issues/855#issuecomment-1069335461

```
esphome:
  name: power-monitor
  platformio_options:
    build_flags: 
      - -DCONFIG_ARDUINO_LOOP_STACK_SIZE=32768

esp32:
  board: nodemcu-32s
  variant: esp32
  framework:
    type: arduino
    version: 2.0.2
    source: https://github.com/espressif/arduino-esp32.git#2.0.2
    platform_version: https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream
```

lumiror, 2022-04-01:
So I tested it, it works. I didn't even have to use the configuration to increase the stack. I only deleted unnecessary sensors, I only have 16 cell batteries. Now I would like to try an older version before 2021, but in the entities I see only entities named ant_bms ..... so only from version 2021 How can I change the name to antbms_old?

lumiror, 2022-04-01:
This is my version of ANT BMS, unfortunately it has no label
[image]

syssi, 2022-04-01:
The label at the photo isn't readable. I'm interesting in the full ....-TB-24S-300A string. Could you provide the first few letters?

lumiror, 2022-04-01:
I tried and it also works but as I wrote in the entities I only see ant_bms from version 2021. But in the log it's right
[image]

syssi, 2022-04-01:
I think this is a feature not a bug. ;-) All sensors have a `unique_id` generated via the MAC address of the ESP plus the sensor key (you cannot change). If you use another ESP for the other BMS everything should be fine. If you want to use one ESP for two BMS you have to load the component twice. I assume in this case the uniqe_id of the second pair of sensors is a bit different.

lumiror, 2022-04-01:
That photo is from aliexpress, here is the original from my BMS.
[image]
[image]

lumiror, 2022-04-01:
I used two ESP32s for each BMS separately but I still see entities only from version 2021

lumiror, 2022-04-01:
I need to change this name. It is possible
[image]
?

syssi, 2022-04-01:
In this case you missed to change the "name:" of the esphome node. The names must be unique per device.

syssi, 2022-04-01:
Please make sure

```
substitutions:
  name: ant-bms.  # <---- to change this name per ESP / ESPHome node.
  external_components_source: github://syssi/esphome-ant-bms@main
  tx_pin: GPIO1
  rx_pin: GPIO3
  password: "12345678"

esphome:
  name: ${name}
  platform: ESP8266
  board: d1_mini
```

lumiror, 2022-04-01:
Of course, I created two names
[image]

syssi, 2022-04-01:
This is strange. Could you provide both config yamls so I can try to reproduce the issue locally. This shouldn't happen.

lumiror, 2022-04-01:
Of course, I created two names:
version 2021:
 substitutions:
  name: ant-bms
  external_components_source: github://syssi/esphome-ant-bms@main
  tx_pin: GPIO17
  rx_pin: GPIO16
  password: "12345678"

version 2020:

substitutions:
  name: antbms-old
  external_components_source: github://syssi/esphome-ant-bms@main
  tx_pin: GPIO17
  rx_pin: GPIO16

lumiror, 2022-04-01:
[2020.txt](https://github.com/syssi/esphome-ant-bms/files/8400527/2020.txt)
[2021.txt](https://github.com/syssi/esphome-ant-bms/files/8400528/2021.txt)

syssi, 2022-04-01:
I think you are mixing things up here becaue the platform/integration can be used multiple times (on the same or other nodes) without trouble. Just make sure the entities are unique. This should be the case if the `$name` is substituted with different values.

Please go to the HA configuration, select "devices & services", select the "devices" tab and search for "bms". Two esphome devices should be listed now. Please click on the device and provide a screenshot of each devices. I'm interested in the different sensor names.

lumiror, 2022-04-01:
I sent you 2x txt file there are also those names. But if you need I can make a screnshot

syssi, 2022-04-01:
The screenshot is important because it looks like there is something missing at you HA setup. Probably you missed to add the second esphome node to HA.

lumiror, 2022-04-01:
One more question, what is the difference between .yaml for version 2021 and before 2021

syssi, 2022-04-01:
The sensors are the same for both devices. Just the controls are a bit different. For example: At the 2021 version there is a switch to enable balancing. At the 2020 version you have to use the balancing button entity. I will try to provide better examples in future so it will be easier to see the differences of the supported control entities.

syssi, 2022-04-02:
The `balancer`, `bluetooth` and `buzzer` switch isn't supported by the old BMS version. If you try to use/toggle these switches you will see an unsupported warning at the logs. Please use the `button.balancer` to start balancing at the old BMS version and `switch.balancer` to control the balancing at the new BMS version.

syssi, 2022-04-02:
Feel free to close this issue as soon your setup is working as expected.

syssi, 2022-04-03:
I close this issue. Feel free to create a new one if you have trouble / something doesn't work as expected / a feature is missing.

closed 2022-04-03 by syssi

jpeletier, 2022-06-18:
@syssi check if [https://github.com/esphome/esphome/pull/3577](https://github.com/esphome/esphome/pull/3577) fixes your issue

syssi, 2022-06-24:
@jpeletier Thanks for your support! I'm following the progress of your pull request. I've seen @SeByDocKy already provided some input about one of his huge ESPHome setups. I cannot add much here. :-)

SeByDocKy, 2022-06-24:
@syssi 

Got a good news for anyone got problem of ESP32's core 1 crash.... Now we can also compile with arduino V2.0..3 from hassio directely via : 

```esphome:
  name: pvbrain
#  platformio_options:
#    build_flags: 
#      - -DCONFIG_ARDUINO_LOOP_STACK_SIZE=32768 #16384 #8192 #16384 # #  
#    platform_packages:
#      - framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#2.0.2
#    platform:
#      - https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream  
      
esp32:
  board: esp32dev
  framework:
    #type: arduino
    #version: latest #2.0.0
    type: arduino
    version: 2.0.3
    platform_version: 4.4.0
    #source: "https://github.com/espressif/arduino-esp32.git#2.0.2"
    #platform_version: >-
    #  https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream  
    
external_components:
  - source: github://syssi/esphome-jk-bms@main
    refresh: 0s
  - source: github://syssi/esphome-pipsolar@pip8048
    refresh: 0s  `````

We also put a githup of our "PVbrain" there : https://github.com/Bandit-17/PVBRAIN/  where you can find how to integrate for example the PIP8048 with new number/select possibilities

Rainha1978, 2023-01-23:
Hello.
First time here, don't know exactly how this work.
Want to leave a question, it's possible with esp32 connect an ANT BMS to inverter?
I want my inverter controll battery based in real values, it's only connected by + and - cables of battery, work's almost good, but not enough.
I have a Goodwee GW5048-ES, can use CAN and UART protocol's.
Thanks for your time.

Regards
Nuno Rainha

syssi, 2023-01-23:
Please create a new issue! Your question is off-topic.