syssi/esphome-ant-bms #37: Power Sensor

lucasimons, 2022-05-01:
there is some issue with the power sensor with 2021 bms version, i see in the app -144w instead the Home Assistant sensor show 9999
[logs_ant-bms_logs.txt](https://github.com/syssi/esphome-ant-bms/files/8598790/logs_ant-bms_logs.txt)
w

syssi, 2022-05-01:
It looks like the BMS doesn't provide a valid `current power` value at register/address 111. A calculated value (total_voltage * current) is provided in this case now.

closed 2022-05-01 by syssi

lucasimons, 2022-05-01:
I try to install new branch but the power stay always 9999w

syssi, 2022-05-01:
Please make sure to disable the new commands:

```
ant_bms:
  id: bms0
  ant_modbus_id: modbus0
  supports_new_commands: false
```

This will enable the special behavior for the 2021 BMS version.

syssi, 2022-05-01:
Oh, wait! 2021 is the new BMS version. I mixed something up here.

syssi, 2022-05-01:
I've pushed a fix. Please try again. :-)

lucasimons, 2022-05-01:
it looks like it hasn't changed yet i always see 9999w

syssi, 2022-05-01:
Could you provide your config yaml?

lucasimons, 2022-05-01:
> Please make sure to disable the new commands:
> 
> ```
> ant_bms:
>   id: bms0
>   ant_modbus_id: modbus0
>   supports_new_commands: false
> ```
> 
> This will enable the special behavior for the 2021 BMS version.

i use the new command because this problem of power value is on a 2021 bms

syssi, 2022-05-01:
This is the new behavior:

https://github.com/syssi/esphome-ant-bms/commit/58de0b20359e6559d66815d44cd44698d911e5d7

In words: If the `supports_new_commands_` boolean is enabled the power will be calculated by voltage * current. If the new commands aren't supported/enabled register 111 from the response is returned.

lucasimons, 2022-05-01:
[logs_ant-bms_logs.txt](https://github.com/syssi/esphome-ant-bms/files/8599527/logs_ant-bms_logs.txt)

`

logger:
  level: VERY_VERBOSE
  logs:
    api.service: WARN
    ota: WARN
    sensor: WARN
    wif: WARN

uart:
  id: uart0
  baud_rate: 19200
  rx_buffer_size: 384
  tx_pin: GPIO16
  rx_pin: GPIO17
  debug:
    direction: BOTH

ant_modbus:
  id: modbus0
  uart_id: uart0

ant_bms:
  id: bms0
  ant_modbus_id: modbus0
  password: "${password}"
  update_interval: 10s
  supports_new_commands: true
  
sensor:
  - platform: ant_bms
    battery_strings:
      name: "${name} battery strings"
    current:
      name: "${name} current"
    soc:
      name: "${name} soc"
    total_battery_capacity_setting:
      name: "${name} total battery capacity setting"
    capacity_remaining:
      name: "${name} capacity remaining"
    battery_cycle_capacity:
      name: "${name} battery cycle capacity"
    total_voltage:
      name: "${name} total voltage"
    total_runtime:
      name: "${name} total runtime"
    average_cell_voltage:
      name: "${name} average cell voltage"
    power:
      name: "${name} power"
    min_cell_voltage:
      name: "${name} min cell voltage"
    max_cell_voltage:
      name: "${name} max cell voltage"
    temperature_1:
      name: "${name} temperature 1"
    temperature_2:
      name: "${name} temperature 2"
    temperature_3:
      name: "${name} temperature 3"
    temperature_4:
      name: "${name} temperature 4"

    cell_voltage_1:
      name: "${name} cell voltage 1"
    cell_voltage_2:
      name: "${name} cell voltage 2"
    cell_voltage_3:
      name: "${name} cell voltage 3"
    cell_voltage_4:
      name: "${name} cell voltage 4"
    cell_voltage_5:
      name: "${name} cell voltage 5"
    cell_voltage_6:
      name: "${name} cell voltage 6"
    cell_voltage_7:
      name: "${name} cell voltage 7"
    cell_voltage_8:
      name: "${name} cell voltage 8"
    charge_mosfet_status_code:
      name: "${name} charge mosfet status code"
    discharge_mosfet_status_code:
      name: "${name} discharge mosfet status code"
    balancer_status_code:
      name: "${name} balancer status code"  

text_sensor:
  - platform: ant_bms
    charge_mosfet_status:
      name: "${name} charge mosfet status"
    discharge_mosfet_status:
      name: "${name} discharge mosfet status"
    balancer_status:
      name: "${name} balancer status"
      
switch:
  - platform: ant_bms
    charging:
      name: "${name} charging"
    discharging:
      name: "${name} discharging"
    balancer:
      name: "${name} balancer"
button:
  - platform: ant_bms
    shutdown:
      name: "${name} shutdown"
    clear_counter:
      name: "${name} clear counter"
    balancer:
      name: "${name} balancer"
    factory_reset:
      name: "${name} factory reset"
    restart:
      name: "${name} restart"  `

syssi, 2022-05-01:
I found the root-cause: You are using the feature branch. I've pushed the fix directly to `main`. Please switch to the `main` branch + compile & flash again.

lucasimons, 2022-05-01:
now it works fine

syssi, 2022-05-01:
Could you do me a favor and verify later the week the power sensor is working properly (positive and negative sign) for both BMS versions?

lucasimons, 2022-05-12:
the power sensor works fine, i measure the current and the bms show more current than my instrument, would be interesting insert a costant to adjust the power, P= V*I*k, k = 0.94 or something else, or adjust the current so I = I(measured by bms) * k

syssi, 2022-05-12:
You can attach a filter to the sensor to manipulate the data.

syssi, 2022-05-12:
```
    power:
      name: "${name} power"
      filters:
        - multiply: 0.94
```

or you could do multiple measurements and use a linear calibration:

```
    power:
      name: "${name} power"
      filters:
        - calibrate_linear:
            - 0.0 -> 0.0
            - 40.0 -> 45.0
            - 100.0 -> 102.5
```

[The left side is the measured value. ](https://esphome.io/components/sensor/index.html#calibrate-linear)

commit 7a12045b (2022-05-01): Provide a calculated power sensor value (2021 BMS version) (Closes: #37) (#38)