syssi/esphome-ant-bms #45: Ant BMS connection to Goodwe inverter.

Rainha1978, 2023-01-23:
Hello.
It's possible to connect Ant BMS to Goodwee inverter with an ESP?

I have a battery with an ANT BMS connected by + and - wires to inverter, don't work in best conditions.
If it's possible, what it's need to put it working?
Inverter is Goodwe GW5048-ES.

Don't have much knowledge about ESP.

Thanks

Nuno Rainha

syssi, 2023-01-23:
Thanks for creating this issue! :-) The component retrieves the BMS status (various values) periodically. These values can be exposed as sensor states. If you find a second component supporting your inverter you could forward the values to the inverter. There is no plug'n'play solution available for your usecase.

Rainha1978, 2023-01-23:
Thanks for responding.

Soo, can't connect inverter by CAN or UART to the ESP to read those values? 
I think the most important value that inverter need is the SOC, because almost all program of the inverter works with that value, and it's the only problem I have this moment, the SOC isn't correct l.
The cycle of charge and discharge, only work's when battery is at 100%, after that the value varieties, and charging/discharging increase that difference.

Rainha1978, 2023-01-23:
Ok.

I leave here some informations about configurations on the application.
Only to know...
[image]
[image]
[image]