syssi/esphome-ant-bms #48: its posible add ble support like jbd-bms? thx.

v222008, 2023-02-22:


syssi, 2023-02-22:
Do you own an ANT-BMS and are you able to use the Android app? Could you provide a btnsoop capture of the bluetooth traffic? This is an article how to create a capture and retrieve it from an android device:

https://medium.com/@charlie.d.anderson/how-to-get-the-bluetooth-host-controller-interface-logs-from-a-modern-android-phone-d23bde00b9fa

syssi, 2023-02-26:
@v222008 Friendly reminder.

syssi, 2023-02-28:
It looks like new ANT-BMS models (f.e. `ANT-BLE16ZMUB`) are using BLE services/characteristics now. This is cool because the previous bluetooth module requiring RFcomm (a serial connection via bluetooth) isn't supported by ESPHome yet.

@RoboDurden did mention the the BLE service (`0000FFE0-0000-1000-8000-00805F9B34FB`) and the two characteristics (`0000FFE1-0000-1000-8000-00805F9B34FB`, `0000FFE2-0000-1000-8000-00805F9B34FB`) [here](https://www.photovoltaikforum.com/thread/183467-wo-neuste-ant-bms-android-app-oder-alternativen-oder-uart-protokoll/?postID=2810989#post2810989).

This looks like the implementation will be very similar to the JBD one: https://github.com/syssi/esphome-jbd-bms/blob/main/components/jbd_bms_ble/jbd_bms_ble.cpp#L10-L12

A btsnoop capture is required to move on.

RoboDurden, 2023-02-28:
Have you asked the Photovoltaik-Forum.com ?

What i need now is a Bluetooth app for the esp32 here in my little solar camper.
I could test this today https://github.com/ArtronShop/ESP32-BLE-Sample

Then we would have GATT connection where we can change whatever we want and capture ble protocol data..
If this would already help you.

syssi, 2023-02-28:
> Have you asked the Photovoltaik-Forum.com ?

I will give it a try!

> What i need now is a Bluetooth app for the esp32 here in my little solar camper.
> I could test this today https://github.com/ArtronShop/ESP32-BLE-Sample

I would like to see the traffic between the Android app and the BMS. So we can simply replicate the traffic. A btsnoop capture can be opened & analyzed using wireshark easily. You don't currently carry a ANT-BMS around with you, do you?

RoboDurden, 2023-02-28:
No: https://youtu.be/hnwDNyYJMKI

v222008, 2023-03-03:
> Do you own an ANT-BMS and are you able to use the Android app? Could you provide a btnsoop capture of the bluetooth traffic? This is an article how to create a capture and retrieve it from an android device:

Thanks for the quick reply. This is how I use ant bms in my ecosystem. I will take your advice and send you the data as soon as possible

v222008, 2023-03-03:
> It looks like new ANT-BMS models (f.e. `ANT-BLE16ZMUB`) are using BLE services/characteristics now. This is cool because the previous bluetooth module requiring RFcomm (a serial connection via [...]
yes it's the ANT-BLE16ZMUB
[image]

RoboDurden, 2023-03-06:
Okay, here my first quick log: https://www.robosoft.de/temp/btsnoop_hci.log
I activated my 16ZM by turning on the hybrid inverter which put a load of about 50 Watt on the bms.
Then the ANT BMS app recognized the bms and showed the cell data.
I also entered the ParameterSet tab and reset to values.

So you now continue and tell us how you did proceed with the btsnoop file :-)

syssi, 2023-03-06:
Good job and welcome back home! The btsnoop file can be opened and analyzed using wireshark. It looks like the MAC address of your ANT-BMS is `16:aa:22:02:23:45`. Let's ask wireshark to filter the capture:

```
bluetooth.addr == 16:aa:22:02:23:45
``` 

- Frame no. 275 to 328: The Android app asks the BMS about supported services & characteristics
- Frame no 331: The Android app subscribes to the `notifications` on handle `0x11`. Handle 0x11 is provided by Service `0xffe0`, Characteristic `0xffe1`
- Frame no 335: The app writes a command to handle `0x10` (Service `0xffe0`, Char `0xffe1`) with the payload `7ea1026c022058c4aa55`. This is the same protocol as the UART one.
- Frame no 337 to 339: The BMS responds with three chunks:

```
7ea1126c022031365a4d00000000000000000000 000031365a4d554230302d323131303236417208 ff0b000041f2aa55
```

- Next command: 7ea1010000be1855aa55
- Response: 

```
7ea11100008e0501021000000000000000008000
8001000000000000000000000000e40ce40ce50c
e50ce80ce70ce70ce60ce80ce70ce70ce70ce70c
e70ce60ce90c0100020002000700a41403005b00
6400010100000076b010d5670e0fba324a000f00
000010582e0200000000e90c1000e40c01000500
e60c000080007a000f02f2fab98c3b00bbd85800
da2d4300e8b649000543aa55
```

- Next command: 7ea1010000be1855aa55
- Repsonse:

```
7ea11100008e0501021000000000000000008000
8001000000000000000000000000e40ce40ce50c
e50ce80ce60ce70ce60ce80ce70ce70ce70ce70c
e70ce60ce80c0100020002000700a41406005b00
6400010100000076b010a8670e0fba324a000f00
000010582e0200000000e80c0500e40c01000400
e60c000080007a000f02f2fab98c3b00bbd85800
db2d4300e8b649002746aa55
```

- Next command: 7ea1010000be1855aa55
- Response: ...cell voltages response...

...

syssi, 2023-03-06:
Some more details about the `7ea1 ... aa55` protocol:

https://github.com/syssi/esphome-ant-bms/issues/20
https://github.com/syssi/esphome-ant-bms/issues/18#issuecomment-1445843701
https://github.com/syssi/esphome-ant-bms/issues/18#issuecomment-1291972132

RoboDurden, 2023-03-06:
Would be awesome if you now could code a ble android app. What is this esphome anyway ? linux app ?
the apk could be this one: https://www.robosoft.de/temp/antbms.apk
Am not sure as i do not remember the original link and there are differen antbms apps around.
Mine has written a 1.3.0 in the Mine->AboutUs tab.

syssi, 2023-03-06:
This is [ESPHome](https://esphome.io): ESPHome is a system to control your ESP8266/ESP32 by simple yet powerful configuration files and control them remotely through Home Automation systems.

RoboDurden, 2023-03-06:
This must be the official download: https://rjxzstech.com/download/ANT-BMS-APP-20211116.html
It contains a mybms_v1.3.0_20211116.apk

syssi, 2023-03-07:
The architecture of ESPHome is bit hard to understand the first time. This short screen capture shows how to install the command line interface, generate a firmware using a YAML file + let ESPHome generate and compile the code + flash the firmware to the device:

[image]

I will try to explain to idea in a few words / bullet points:

1. If you want to retrieve some measurements from your ANT-BMS using UART-TTL you could pick a ESP32 or ESP8266 and attach the ANT-BMS to some GPIOs
2. This YAML describes an example configuration: https://github.com/syssi/esphome-ant-bms/blob/main/esp8266-example.yaml
3. Each block at the YAML is called "component". You can find some documentation per component on esphome.io. If you load the `wifi` component the ESP is able to connect to your local network. If you load the `mqtt` component the ESP is able to publish measurements/sensor states using MQTT and so on. If you remove components you will remove features of the ESPHome node.

Let's have a look at the example configuration of this project:

```
# Populate some variables/keys with values
substitutions:
  name: ant-bms

# Load the esphome component and set the hostname 
esphome:
  name: ${name}

# Load the esp8266 platform and set the platformio board name. Now you could use the consts `D1, D2, ..., Dx, A0` etc. instead of `GPIOx`.
esp8266:
  board: d1_mini

# Clone the main branch of my external component to make the `ant_bms` component available.
# This `ant_bms` component isn't part of the ESPHome core so it must be retrieved here.
external_components:
  - source: github://syssi/esphome-ant-bms@main
    refresh: 0s

# Setup WiFi and connect to your wireless network using the key `wifi_ssid` and `wifi_password` from the secrets.yaml
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

# Load the `ota` component to allow OTA updates
ota:

# Load the `logger`. It writes to GPIO1/GPIO3 per default. The logs are accessible/available via `mqtt`/`api` too. So you don't need to attach a USB-to-TTL converter if you want to see some logs
logger:

# If you use Home Assistant please remove this `mqtt` section and uncomment the native `api` component!
# api:

# Load the mqtt component to publish all states/measurements to a MQTT broker
mqtt:
  broker: !secret mqtt_host
  username: !secret mqtt_username
  password: !secret mqtt_password
  id: mqtt_client

# Attach the `uart` component to GPIO4+GPIO5 and set the baud rate
uart:
  - id: uart0
    baud_rate: 19200
    rx_buffer_size: 384
    tx_pin: GPIO4
    rx_pin: GPIO5

# Wire the `ant_bms` component to the `uart` component and call the `update` method every 10 seconds.
# On every update_interval this implements sends a `cell info request` to the BMS. If the BMS responds the
# response will be decoded and sensors/states are updated.
ant_bms:
  - id: bms0
    uart_id: uart0
    update_interval: 10s
    rx_timeout: 50ms
    supports_new_commands: true
    password: "12345678"

# Expose some measurements as sensors
sensor:
  - platform: ant_bms
    ant_bms_id: bms0
    soc:
      name: "${name} soc"
    total_voltage:
      name: "${name} total voltage"

# Expose some strings as text_sensors
text_sensor:
  - platform: ant_bms
    ant_bms_id: bms0
    total_runtime_formatted:
      name: "${name} total runtime formatted"

# Expose some controls (writable registers) as `switch` entity and send a command to the BMS if you toggle the switch
switch:
  - platform: ant_bms
    ant_bms_id: bms0
    charging:
      name: "${name} charging"
    discharging:
      name: "${name} discharging"

# Create a button entity to restart the BMS
button:
  - platform: ant_bms
    ant_bms_id: bms0
    restart:
      name: "${name} restart"
```

My project implements the `ant_bms` platform.

syssi, 2023-03-09:
A first draft of the bluetooth support is ready. Please use this YAML configuration for testing:

https://raw.githubusercontent.com/syssi/esphome-ant-bms/add-ble-support/esp32-ble-example.yaml

syssi, 2023-03-09:
> But when everybody is eating shit you better eat is as well :-/

Millions of flies can not be wrong. :-P

v222008, 2023-03-09:
> Do you own an ANT-BMS and are you able to use the Android app? Could you provide a btnsoop capture of the bluetooth traffic? This is an article how to create a capture and retrieve it from an [...]
sorry for the delay, I use an iPhone, I have an Android phone with Android version 5. managed to enter and pair with BMS and get telemetry data

[btsnoop_hci(1).log](https://github.com/syssi/esphome-ant-bms/files/10933421/btsnoop_hci.1.log)

syssi, 2023-03-09:
@v222008 Could you use this YAML configuration, update the `mac_address` substitution to the MAC address of your device and flash it to an ESP32? I would be happy about the log output. It should include some raw traffic.

https://raw.githubusercontent.com/syssi/esphome-ant-bms/add-ble-support/esp32-ble-example.yaml

RoboDurden, 2023-03-09:
Okay `esphome run esp8266-example.yaml ` succeed and flashed onto a D1 Mini :-)

but not `esphome run esp32-ble-example.yaml`:
```
Installing collected packages: pyparsing, msgpack, lockfile, urllib3, six, pyyaml, pycparser, packaging, idna, future, contextlib2, colorama, charset-normalizer, certifi, tqdm, schema, requests, click, cffi, requests-toolbelt, requests-file, cryptography, cachecontrol, idf-component-manager
  Attempting uninstall: urllib3
    Found existing installation: urllib3 1.26.14
    Uninstalling urllib3-1.26.14:
      Successfully uninstalled urllib3-1.26.14
  Attempting uninstall: six
    Found existing installation: six 1.16.0
    Uninstalling six-1.16.0:
      Successfully uninstalled six-1.16.0
  Attempting uninstall: pyyaml
    Found existing installation: PyYAML 6.0
    Uninstalling PyYAML-6.0:
      Successfully uninstalled PyYAML-6.0
ERROR: Could not install packages due to an OSError: [WinError 5] Zugriff verweigert: 'C:\\Users\\PAN CF-LX6\\AppData\\Local\\Programs\\Python\\Python311\\Lib\\site-packages\\~aml\\_yaml.cp311-win_amd64.pyd'
Consider using the `--user` option or check the permissions.

[notice] A new release of pip available: 22.3.1 -> 23.0.1
[notice] To update, run: python.exe -m pip install --upgrade pip
*** Error 1
Installing windows-curses package
Der Befehl "C:\Users\PAN" ist entweder falsch geschrieben oder
konnte nicht gefunden werden.
*** Error 1
Error: Detected a whitespace character in project paths.
============================================= [FAILED] Took 392.96 seconds =============================================

c:\EspHome\esphome-seplos-bms>

```

typical linux command line crab.

syssi, 2023-03-09:
Sorry I cannot help here. I don't use Windows. :-)

v222008, 2023-03-09:
> @v222008 Could you use this YAML configuration, update the `mac_address` substitution to the MAC address of your device and flash it to an ESP32? I would be happy about the log output. It should [...]
ok its works)
[logs_ant-bms-6kwt_run (1).txt](https://github.com/syssi/esphome-ant-bms/files/10933650/logs_ant-bms-6kwt_run.1.txt)

syssi, 2023-03-09:
Perfect! Some details doesn't look good but I know what's going on. I will prepare a fix.

syssi, 2023-03-09:
Your cell count (14S vs. 16S) is repected now. Please flash again and provide another log.

Are these values plausible?

```
[18:21:28][I][ant_bms_ble:354]:   Battery capacity: 150.00 Ah
[18:21:28][I][ant_bms_ble:357]:   Battery capacity remaining: 6.80 Ah
[18:21:28][I][ant_bms_ble:360]:   Battery cycles capacity: 3010.07 Ah
[18:21:28][I][ant_bms_ble:366]:   Total runtime: 109d 20h
[18:21:28][I][ant_bms_ble:396]:   Accumulated discharging capacity: 3119470 Ah?
[18:21:28][I][ant_bms_ble:399]:   Accumulated charging capacity: 2900663 Ah?
[18:21:28][I][ant_bms_ble:402]:   Accumulated discharging time: 9d 8h
[18:21:28][I][ant_bms_ble:405]:   Accumulated charging time: 7d 19h
```

I assume it should be:

```
[18:21:28][I][ant_bms_ble:354]:   Battery capacity: 150.00 Ah
[18:21:28][I][ant_bms_ble:357]:   Battery capacity remaining: 6.80 Ah
[18:21:28][I][ant_bms_ble:360]:   Battery cycles capacity: 3010.07 Ah
[18:21:28][I][ant_bms_ble:366]:   Total runtime: 109d 20h
[18:21:28][I][ant_bms_ble:396]:   Accumulated discharging capacity: 3119.470 Ah
[18:21:28][I][ant_bms_ble:399]:   Accumulated charging capacity: 2900.663 Ah
[18:21:28][I][ant_bms_ble:402]:   Accumulated discharging time: 9d 8h
[18:21:28][I][ant_bms_ble:405]:   Accumulated charging time: 7d 19h
```

Correct? Did you charge your battery around 20 times (3010/150=20,06) in total?

v222008, 2023-03-09:
> cell count (14S

cell count  14S 
[logs_ant-bms-6kwt_logs_last.txt](https://github.com/syssi/esphome-ant-bms/files/10934430/logs_ant-bms-6kwt_logs_last.txt)

syssi, 2023-03-09:
Most of the sensor entities are available now.

v222008, 2023-03-10:
> Most of the sensor entities are available now.

great job thank you very much.

v222008, 2023-03-10:
> Most of the sensor entities are available now.

Аre you planning to start supporting BMS smart board DaLy ?

syssi, 2023-03-10:
I don't understand your question. Do you ask about another [device](https://esphome.io/components/sensor/daly_bms.html)?

syssi, 2023-03-16:
I've merged the BLE support into `main`. Please create a new issue if something doesn't work as expected.

closed 2023-03-16 by syssi

kirillch2000, 2024-10-01:
> > I don't understand your question. Do you ask about another [device](https://esphome.io/components/sensor/daly_bms.html)?
> 
> I apologize for writing about another device in the wrong topic because I did not find it in your repositories. I am certainly interested in support BLE

Yes, I am also very interested in supporting DALY BMS. And I also could not find this popular BMS in your projects for ESPHOME :(

syssi, 2024-11-18:
@kirillch2000 See https://github.com/syssi/esphome-daly-bms