syssi/esphome-ant-bms #67: Problem with update

lumiror, 2023-05-01:
Hello, I wanted to update the compilation after installing the new homassistant version, but it is not possible because I have this error. Can you please advise how to fix it. Thank you

[image]

syssi, 2023-05-01:
Please rename `uart0` to `uart_0`. This should solve the validation issue.

lumiror, 2023-05-01:
I tried that, but it didn't work

[image]

syssi, 2023-05-02:
Please remove the `ant_modbus` section and update the `ant_bms` section to:

```
uart:
  - id: uart_0
    baud_rate: 19200
    rx_buffer_size: 384
    tx_pin: ${tx_pin}
    rx_pin: ${rx_pin}

ant_bms:
  - id: bms0
    uart_id: uart_0
    update_interval: 5s
    rx_timeout: ${rx_timeout}
    supports_new_commands: false
```

lumiror, 2023-05-02:
something is still wrong, please take a look

[image]

syssi, 2023-05-02:
Please remove line 54 (`rx_timeout: ...`) and try again.

lumiror, 2023-05-02:
and line 55 should be true or false because before it was true and now it is false?

lumiror, 2023-05-02:
and again, please, in the new version after 2021, it's like this, what should I delete there.

[image]

lumiror, 2023-05-02:
can it be like this
[image]

syssi, 2023-05-02:
It's pretty the same here but `supports_new_commands: true`:

```
uart:
  - id: uart_0
    baud_rate: 19200
    rx_buffer_size: 384
    tx_pin: ${tx_pin}
    rx_pin: ${rx_pin}

ant_bms:
  - id: bms0
    uart_id: uart_0
    update_interval: 5s
    password: "${password}"
    supports_new_commands: true
```

lumiror, 2023-05-02:
so I understood it correctly.
but why is it true in one and false in the other?

syssi, 2023-05-02:
Probably because of the different hardware/software version of the BMS?

closed 2023-06-12 by syssi