syssi/esphome-ant-bms #68: 'ant-bms current': Sending state nan A with 1 decimals of accuracy

furang, 2023-06-11:
Board: D1 mini
Ant bms S Version: 16ZMUB00-220825A 
Ant bms H Version: 16ZMA0TB080A
Battery configuration: LiFePO4 8S
Config:
```yaml
substitutions:
  name: ant-bms
  device_description: "Monitor and control a ANT Battery Management System (ANT-BMS) via UART"
  external_components_source: github://syssi/esphome-ant-bms@main
  rx_timeout: 50ms
  tx_pin: GPIO4
  rx_pin: GPIO5
  password: "12345678"

esphome:
  name: ${name}
  comment: ${device_description}
  project:
    name: "syssi.esphome-ant-bms"
    version: 2.0.0

esp8266:
  board: d1_mini

external_components:
  - source: ${external_components_source}
    refresh: 0s

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "API KEY"

ota:
  password: "OTA KEY"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Antbms Fallback Hotspot"
    password: "pass"

#captive_portal:
    
#mqtt:
#  broker: !secret mqtt_host
#  username: !secret mqtt_username
#  password: !secret mqtt_password
#  id: mqtt_client

uart:
  - id: uart_0
    baud_rate: 19200
    rx_buffer_size: 384
    tx_pin: ${tx_pin}
    rx_pin: ${rx_pin}

ant_bms:
  - id: bms0
    uart_id: uart_0
    update_interval: 10s
    rx_timeout: ${rx_timeout}
    supports_new_commands: true
    password: "${password}"

binary_sensor:
  - platform: ant_bms
    ant_bms_id: bms0
    online_status:
      name: "${name} online status"

sensor:
  - platform: ant_bms
    ant_bms_id: bms0
    battery_strings:
      name: "${name} battery strings"
    current:
      name: "${name} current"
    soc:
      name: "${name} soc"
    total_battery_capacity_setting:
      name: "${name} total battery capacity setting"
    capacity_remaining:
      name: "${name} capacity remaining"
    battery_cycle_capacity:
      name: "${name} battery cycle capacity"
    total_voltage:
      name: "${name} total voltage"
    total_runtime:
      name: "${name} total runtime"
    average_cell_voltage:
      name: "${name} average cell voltage"
    power:
      name: "${name} power"
    min_cell_voltage:
      name: "${name} min cell voltage"
    max_cell_voltage:
      name: "${name} max cell voltage"
    min_voltage_cell:
      name: "${name} min voltage cell"
    max_voltage_cell:
      name: "${name} max voltage cell"
    delta_cell_voltage:
      name: "{name} delta cell voltage"
    temperature_1:
      name: "${name} temperature 1"
    temperature_2:
      name: "${name} temperature 2"
    temperature_3:
      name: "${name} temperature 3"
    temperature_4:
      name: "${name} temperature 4"
    cell_voltage_1:
      name: "${name} cell voltage 1"
    cell_voltage_2:
      name: "${name} cell voltage 2"
    cell_voltage_3:
      name: "${name} cell voltage 3"
    cell_voltage_4:
      name: "${name} cell voltage 4"
    cell_voltage_5:
      name: "${name} cell voltage 5"
    cell_voltage_6:
      name: "${name} cell voltage 6"
    cell_voltage_7:
      name: "${name} cell voltage 7"
    cell_voltage_8:
      name: "${name} cell voltage 8"
    charge_mosfet_status_code:
      name: "${name} charge mosfet status code"
    discharge_mosfet_status_code:
      name: "${name} discharge mosfet status code"
    balancer_status_code:
      name: "${name} balancer status code"

text_sensor:
  - platform: ant_bms
    ant_bms_id: bms0
    charge_mosfet_status:
      name: "${name} charge mosfet status"
    discharge_mosfet_status:
      name: "${name} discharge mosfet status"
    balancer_status:
      name: "${name} balancer status"
    total_runtime_formatted:
      name: "${name} total runtime formatted"

switch:
  - platform: ant_bms
    ant_bms_id: bms0
    charging:
      name: "${name} charging"
    discharging:
      name: "${name} discharging"
    # BMS version 2021 only
    # Please enable supports_new_commands
    balancer:
      name: "${name} balancer"

button:
  - platform: ant_bms
    ant_bms_id: bms0
    shutdown:
      name: "${name} shutdown"
    clear_counter:
      name: "${name} clear counter"
    balancer:
      name: "${name} balancer"
    factory_reset:
      name: "${name} factory reset"
    restart:
      name: "${name} restart"
```
Logs:
```
[14:41:03][D][binary_sensor:036]: 'ant-bms online status': Sending state OFF
[14:41:03][D][text_sensor:064]: 'ant-bms discharge mosfet status': Sending state 'Offline'
[14:41:03][D][text_sensor:064]: 'ant-bms charge mosfet status': Sending state 'Offline'
[14:41:03][D][text_sensor:064]: 'ant-bms balancer status': Sending state 'Offline'
[14:41:03][D][text_sensor:064]: 'ant-bms total runtime formatted': Sending state 'Offline'
[14:41:03][D][sensor:093]: 'ant-bms battery strings': Sending state nan  with 0 decimals of accuracy
[14:41:03][D][sensor:093]: 'ant-bms current': Sending state nan A with 1 decimals of accuracy
[14:41:03][D][sensor:093]: 'ant-bms soc': Sending state nan % with 0 decimals of accuracy
[14:41:03][D][sensor:093]: 'ant-bms total battery capacity setting': Sending state nan Ah with 2 decimals of accuracy
[14:41:03][D][sensor:093]: 'ant-bms capacity remaining': Sending state nan Ah with 2 decimals of accuracy
[14:41:03][D][sensor:093]: 'ant-bms battery cycle capacity': Sending state nan Ah with 2 decimals of accuracy
[14:41:03][D][sensor:093]: 'ant-bms total voltage': Sending state nan V with 2 decimals of accuracy
[14:41:03][D][sensor:093]: 'ant-bms total runtime': Sending state nan s with 0 decimals of accuracy
[14:41:03][D][sensor:093]: 'ant-bms average cell voltage': Sending state nan V with 2 decimals of accuracy
[14:41:03][D][sensor:093]: 'ant-bms power': Sending state nan W with 2 decimals of accuracy
[14:41:03][D][sensor:093]: 'ant-bms min cell voltage': Sending state nan V with 3 decimals of accuracy
[14:41:03][D][sensor:093]: 'ant-bms max cell voltage': Sending state nan V with 3 decimals of accuracy
[14:41:03][D][sensor:093]: 'ant-bms min voltage cell': Sending state nan  with 0 decimals of accuracy
[14:41:03][D][sensor:093]: 'ant-bms max voltage cell': Sending state nan  with 0 decimals of accuracy
[14:41:03][D][sensor:093]: 'ant-bms charge mosfet status code': Sending state nan  with 0 decimals of accuracy
[14:41:03][D][sensor:093]: 'ant-bms discharge mosfet status code': Sending state nan  with 0 decimals of accuracy
[14:41:03][D][sensor:093]: 'ant-bms balancer status code': Sending state nan  with 0 decimals of accuracy
[14:41:03][D][sensor:093]: 'ant-bms temperature 1': Sending state nan °C with 0 decimals of accuracy
[14:41:03][D][sensor:093]: 'ant-bms temperature 2': Sending state nan °C with 0 decimals of accuracy
[14:41:03][D][sensor:093]: 'ant-bms temperature 3': Sending state nan °C with 0 decimals of accuracy
[14:41:03][D][sensor:093]: 'ant-bms temperature 4': Sending state nan °C with 0 decimals of accuracy
[14:41:03][D][sensor:093]: 'ant-bms cell voltage 1': Sending state nan V with 3 decimals of accuracy
[14:41:03][D][sensor:093]: 'ant-bms cell voltage 2': Sending state nan V with 3 decimals of accuracy
[14:41:03][D][sensor:093]: 'ant-bms cell voltage 3': Sending state nan V with 3 decimals of accuracy
[14:41:03][D][sensor:093]: 'ant-bms cell voltage 4': Sending state nan V with 3 decimals of accuracy
[14:41:03][D][sensor:093]: 'ant-bms cell voltage 5': Sending state nan V with 3 decimals of accuracy
[14:41:03][D][sensor:093]: 'ant-bms cell voltage 6': Sending state nan V with 3 decimals of accuracy
[14:41:03][D][sensor:093]: 'ant-bms cell voltage 7': Sending state nan V with 3 decimals of accuracy
[14:41:03][D][sensor:093]: 'ant-bms cell voltage 8': Sending state nan V with 3 decimals of accuracy
```

I soldered to pins inside the display.
I also tried flashing with 4 and 5 pin vice versa.
Anyway I get nan instead of values.
What information should I provide additionally?
Thanks!

syssi, 2023-06-11:
Could you flash this YAML just for testing and provide another log capture: https://github.com/syssi/esphome-ant-bms/blob/main/tests/esp8266-query-data-v2021.yaml

This is the background: The `ant_bms` implementation uses the old protocol to retrieve the status frame. See:

https://github.com/syssi/esphome-ant-bms/blob/main/components/ant_bms/ant_bms.cpp#L487-L498

In the past the old and the new BMS firmware was able to respond to the old request but I'm aware of some recent models which respond to the new protocol only. The mentioned YAML above polls the BMS using the status request of the new protocol. If you are lucky your BMS will respond this time.

syssi, 2023-06-11:
Did you attach the TX/RX lines to GPIO4/5 for sure? Could you provide a photo of your wiring?

furang, 2023-06-11:
Sure. I'll try.
[image]
Black case is the lcd screen. I solder up to it's pins. Those wires are connected to ANT BMS. Tape wire is connected to D1 mini (the order is the same). LCD works fine btw.
[image]
This is how it's soldered.
[image]
And here's another side to see soldered pins.

Here's my bms: https://www.aliexpress.com/item/4001202028077.html
7s-16s with LCD

syssi, 2023-06-11:
Please unplug the display an try again. It's possible the display pulls one of the lines down/up and makes the ESP deaf.

furang, 2023-06-13:
I tried to connect it without display. Still no response

```
[23:08:39][C][wifi:505]: WiFi:
[23:08:39][C][wifi:363]:   Local MAC: 8C:AA:B5:CF:27:46
[23:08:39][C][wifi:364]:   SSID: 'Sweet Home 2'[redacted]
[23:08:39][C][wifi:365]:   IP Address: 192.168.88.55
[23:08:39][C][wifi:366]:   BSSID: 18:FD:74:34:E6:3D[redacted]
[23:08:39][C][wifi:368]:   Hostname: 'ant-bms'
[23:08:39][C][wifi:370]:   Signal strength: -47 dB ▂▄▆█
[23:08:39][C][wifi:374]:   Channel: 1
[23:08:39][C][wifi:375]:   Subnet: 255.255.255.0
[23:08:39][C][wifi:376]:   Gateway: 192.168.88.1
[23:08:39][C][wifi:377]:   DNS1: 192.168.88.101
[23:08:39][C][wifi:378]:   DNS2: 0.0.0.0
[23:08:39][C][logger:301]: Logger:
[23:08:39][C][logger:302]:   Level: DEBUG
[23:08:39][C][logger:303]:   Log Baud Rate: 115200
[23:08:39][C][logger:305]:   Hardware UART: UART0
[23:08:39][C][uart.arduino_esp8266:102]: UART Bus:
[23:08:39][C][uart.arduino_esp8266:103]:   TX Pin: GPIO4
[23:08:39][C][uart.arduino_esp8266:104]:   RX Pin: GPIO5
[23:08:39][C][uart.arduino_esp8266:106]:   RX Buffer Size: 256
[23:08:39][C][uart.arduino_esp8266:108]:   Baud Rate: 19200 baud
[23:08:39][C][uart.arduino_esp8266:109]:   Data Bits: 8
[23:08:39][C][uart.arduino_esp8266:110]:   Parity: NONE
[23:08:39][C][uart.arduino_esp8266:111]:   Stop bits: 1
[23:08:39][C][uart.arduino_esp8266:115]:   Using software serial
[23:08:39][C][mdns:108]: mDNS:
[23:08:39][C][mdns:109]:   Hostname: ant-bms
[23:08:39][C][ota:093]: Over-The-Air Updates:
[23:08:39][C][ota:094]:   Address: ant-bms.local:8266
[23:08:39][C][ota:097]:   Using Password.
[23:08:39][C][api:138]: API Server:
[23:08:39][C][api:139]:   Address: ant-bms.local:6053
[23:08:39][C][api:141]:   Using noise encryption: YES
[23:08:40][D][uart_debug:114]: >>> 7E:A1:01:00:00:BE:18:55:AA:55
[23:08:45][D][uart_debug:114]: >>> 7E:A1:01:00:00:BE:18:55:AA:55
[23:08:50][D][uart_debug:114]: >>> 7E:A1:01:00:00:BE:18:55:AA:55
[23:08:55][D][uart_debug:114]: >>> 7E:A1:01:00:00:BE:18:55:AA:55
[23:09:00][D][uart_debug:114]: >>> 7E:A1:01:00:00:BE:18:55:AA:55
[23:09:05][D][uart_debug:114]: >>> 7E:A1:01:00:00:BE:18:55:AA:55
[23:09:10][D][uart_debug:114]: >>> 7E:A1:01:00:00:BE:18:55:AA:55
[23:09:15][D][uart_debug:114]: >>> 7E:A1:01:00:00:BE:18:55:AA:55
[23:09:20][D][uart_debug:114]: >>> 7E:A1:01:00:00:BE:18:55:AA:55
[23:09:25][D][uart_debug:114]: >>> 7E:A1:01:00:00:BE:18:55:AA:55
[23:09:30][D][uart_debug:114]: >>> 7E:A1:01:00:00:BE:18:55:AA:55
[23:09:35][D][uart_debug:114]: >>> 7E:A1:01:00:00:BE:18:55:AA:55
[23:09:40][D][uart_debug:114]: >>> 7E:A1:01:00:00:BE:18:55:AA:55
```

also tried with swaped rx/tx pins. No response.

syssi, 2023-06-13:
Could you try to use the `ant_bms` component again to send the old request payload? If this doesn't work we could try to sniff the traffic between the BMS and the display without polling.

furang, 2023-06-13:
Nothing changed. nans instead of values. I have usb<->uart converter and will try to catch some traffic. But only tomorrow. You can leave some additional instructions what exactly should I do.

furang, 2023-06-18:
[antbms.txt](https://github.com/syssi/esphome-ant-bms/files/11781381/antbms.txt)
Hello. Is this enough?
It seems like baud rate is 9600. Requesting data is just "c" (few packets in the end).
The rest is data itself. I tried switch on and off chg and dischg to generate some additional data.

syssi, 2023-06-18:
How did you identify the baud rate?

furang, 2023-06-18:
I tried multiple variants and watched both sending and receiving traffic. And judging by data (request one) it's 9600 because it's only "c" char. If you try 19200 it's "ZZ".

syssi, 2023-06-18:
Could you provide another capture using 19200 baud? Please don't expect decodable ASCII values. The ANT-BMS protocol is binary.

syssi, 2023-06-19:
Are you able to control your BMS using an Android app? Could you provide a download link?

furang, 2023-06-19:
I'm using iOS App and it works fine.
Here's the link to app store (sorry I don't know how to download package): https://apps.apple.com/ua/app/ant-bms/id1639257625?l=uk
[antbms-res.txt](https://github.com/syssi/esphome-ant-bms/files/11788975/antbms-res.txt)
[antbms-req.txt](https://github.com/syssi/esphome-ant-bms/files/11788976/antbms-req.txt)
And here are files for request and response.

syssi, 2023-06-19:
This capture looks much better. I've extracted these status responses:

```
AA 55 AA FF 01 11 0D 57 0D 59 0D 47 0D 59 0D 5E 0D 5B 0D 5D 0D 6B 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 FF FF FF FF 5C 0D B5 85 80 0C 63 70 B6 00 01 5C DA 00 0F C9 4B 00 1C 00 1B 00 1A 00 1A FF D8 FF D8 01 01 00 00 00 00 00 00 00 00 00 00 08 0D 6B 03 0D 47 0D 5A 08 00 00 00 87 00 83 02 0E 00 00 00 00 00 00 13 D4
AA 55 AA FF 01 11 0D 57 0D 59 0D 47 0D 59 0D 5E 0D 5B 0D 5D 0D 6B 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 FF FF FF FE 5C 0D B5 85 80 0C 63 70 B6 00 01 5C DA 00 0F C9 4B 00 1C 00 1B 00 1A 00 1A FF D8 FF D8 01 01 00 00 00 00 00 00 FF FF FF FE 08 0D 6B 03 0D 47 0D 5A 08 00 00 00 87 00 83 02 0E 00 00 00 00 00 00 17 CE
AA 55 AA FF 01 11 0D 57 0D 59 0D 47 0D 59 0D 5E 0D 5B 0D 5D 0D 6B 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 FF FF FF FF 5C 0D B5 85 80 0C 63 70 B6 00 01 5C DA 00 0F C9 4B 00 1C 00 1B 00 1A 00 1A FF D8 FF D8 01 01 00 00 00 00 00 00 FF FF FF FB 08 0D 6B 03 0D 47 0D 5A 08 00 00 00 87 00 83 02 0E 00 00 00 00 00 00 17 CC
AA 55 AA FF 01 11 0D 57 0D 59 0D 47 0D 59 0D 5E 0D 5B 0D 5D 0D 6B 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 FF FF FF FF 5C 0D B5 85 80 0C 63 70 B6 00 01 5C DA 00 0F C9 4B 00 1C 00 1B 00 1A 00 1A FF D8 FF D8 01 01 00 00 00 00 00 00 FF FF FF FE 08 0D 6B 03 0D 47 0D 5A 08 00 00 00 87 00 83 02 0E 00 00 00 00 00 00 17 CF
AA 55 AA FF 01 11 0D 57 0D 59 0D 47 0D 59 0D 5E 0D 5B 0D 5D 0D 6B 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 5C 0D B5 85 80 0C 63 70 B6 00 01 5C DA 00 0F C9 4C 00 1C 00 1B 00 1A 00 1A FF D8 FF D8 01 01 00 00 00 00 00 00 FF FF FF FE 08 0D 6B 03 0D 47 0D 5A 08 00 00 00 87 00 82 02 0E 00 00 00 00 00 00 13 D3
AA 55 AA FF 01 11 0D 57 0D 59 0D 47 0D 59 0D 5E 0D 5B 0D 5D 0D 6B 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 FF FF FF FE 5C 0D B5 85 80 0C 63 70 B6 00 01 5C DA 00 0F C9 4C 00 1C 00 1B 00 1A 00 1A FF D8 FF D8 01 01 00 00 00 00 00 00 00 00 00 00 08 0D 6B 03 0D 47 0D 5A 08 00 00 00 87 00 82 02 0E 00 00 00 00 00 00 13 D3
[...]
AA 55 AA FF 01 12 0D 57 0D 5A 0D 47 0D 5A 0D 5F 0D 5C 0D 5D 0D 6D 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 5C 0D B5 85 80 0C 63 92 2E 00 01 5C DE 00 0F C9 AE 00 1C 00 1B 00 1A 00 1A FF D8 FF D8 01 01 00 00 00 00 00 00 00 00 00 00 08 0D 6D 03 0D 47 0D 5A 08 00 00 00 87 00 83 02 0E 00 00 00 00 00 00 0F E2
AA 55 AA FF 01 12 0D 57 0D 5A 0D 47 0D 5A 0D 5F 0D 5C 0D 5D 0D 6D 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 5C 0D B5 85 80 0C 63 92 2E 00 01 5C DE 00 0F C9 AE 00 1C 00 1B 00 1A 00 1A FF D8 FF D8 01 01 00 00 00 00 00 00 00 00 00 00 08 0D 6D 03 0D 47 0D 5A 08 00 00 00 87 00 83 02 0E 00 00 00 00 00 00 0F E2
AA 55 AA FF 01 12 0D 57 0D 5A 0D 47 0D 5A 0D 5F 0D 5C 0D 5D 0D 6D 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 5C 0D B5 85 80 0C 63 92 2E 00 01 5C DE 00 0F C9 AE 00 1C 00 1B 00 1A 00 1A FF D8 FF D8 01 01 00 00 00 00 00 00 00 00 00 00 08 0D 6D 03 0D 47 0D 5A 08 00 00 00 87 00 83 02 0E 00 00 00 00 00 00 0F E2
```

syssi, 2023-06-19:
If I pass one of the captured status responses into the component:

```
$ cat esp8266-example-faker.yaml 
<<: !include esp8266-example-debug.yaml

interval:
  - interval: 2s
    then:
      - lambda: |-
          id(bms0).on_ant_bms_data({
            0xAA, 0x55, 0xAA, 0xFF, 0x01, 0x11, 0x0D, 0x57, 0x0D, 0x59, 0x0D, 0x47, 0x0D, 0x59, 0x0D, 0x5E, 0x0D, 0x5B,
            0x0D, 0x5D, 0x0D, 0x6B, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
            0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
            0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0xFF, 0xFF,
            0xFF, 0xFF, 0x5C, 0x0D, 0xB5, 0x85, 0x80, 0x0C, 0x63, 0x70, 0xB6, 0x00, 0x01, 0x5C, 0xDA, 0x00, 0x0F, 0xC9,
            0x4B, 0x00, 0x1C, 0x00, 0x1B, 0x00, 0x1A, 0x00, 0x1A, 0xFF, 0xD8, 0xFF, 0xD8, 0x01, 0x01, 0x00, 0x00, 0x00,
            0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x08, 0x0D, 0x6B, 0x03, 0x0D, 0x47, 0x0D, 0x5A, 0x08, 0x00, 0x00,
            0x00, 0x87, 0x00, 0x83, 0x02, 0x0E, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x13, 0xD4
          });
```

The implementation is able to decode the frame:

```
16:23:54][I][ant_bms:177]: Status frame received
[16:23:54][D][sensor:093]: 'ant-bms total voltage': Sending state 27.30000 V with 2 decimals of accuracy
[16:23:54][D][sensor:093]: 'ant-bms cell voltage 1': Sending state 3.41500 V with 3 decimals of accuracy
[16:23:54][D][sensor:093]: 'ant-bms cell voltage 2': Sending state 3.41700 V with 3 decimals of accuracy
[16:23:54][D][sensor:093]: 'ant-bms cell voltage 3': Sending state 3.39900 V with 3 decimals of accuracy
[16:23:54][D][sensor:093]: 'ant-bms cell voltage 4': Sending state 3.41700 V with 3 decimals of accuracy
[16:23:54][D][sensor:093]: 'ant-bms cell voltage 5': Sending state 3.42200 V with 3 decimals of accuracy
[16:23:54][D][sensor:093]: 'ant-bms cell voltage 6': Sending state 3.41900 V with 3 decimals of accuracy
[16:23:54][D][sensor:093]: 'ant-bms cell voltage 7': Sending state 3.42100 V with 3 decimals of accuracy
[16:23:54][D][sensor:093]: 'ant-bms cell voltage 8': Sending state 3.43500 V with 3 decimals of accuracy
[16:23:54][D][sensor:093]: 'ant-bms current': Sending state -0.10000 A with 1 decimals of accuracy
[16:23:54][D][sensor:093]: 'ant-bms soc': Sending state 92.00000 % with 0 decimals of accuracy
[16:23:54][D][sensor:093]: 'ant-bms total battery capacity setting': Sending state 230.00000 Ah with 2 decimals of accuracy
[16:23:54][D][sensor:093]: 'ant-bms capacity remaining': Sending state 207.84351 Ah with 2 decimals of accuracy
[16:23:54][D][sensor:093]: 'ant-bms battery cycle capacity': Sending state 89.30601 Ah with 2 decimals of accuracy
[16:23:54][D][sensor:093]: 'ant-bms total runtime': Sending state 1034571.00000 s with 0 decimals of accuracy
[16:23:54][D][text_sensor:064]: 'ant-bms total runtime formatted': Sending state '11d 23h'
[16:23:54][D][sensor:093]: 'ant-bms temperature 1': Sending state 28.00000 °C with 0 decimals of accuracy
[16:23:54][D][sensor:093]: 'ant-bms temperature 2': Sending state 27.00000 °C with 0 decimals of accuracy
[16:23:54][D][sensor:093]: 'ant-bms temperature 3': Sending state 26.00000 °C with 0 decimals of accuracy
[16:23:54][D][sensor:093]: 'ant-bms temperature 4': Sending state 26.00000 °C with 0 decimals of accuracy
[16:23:54][D][sensor:093]: 'ant-bms temperature 5': Sending state -40.00000 °C with 0 decimals of accuracy
[16:23:54][D][sensor:093]: 'ant-bms temperature 6': Sending state -40.00000 °C with 0 decimals of accuracy
[16:23:54][D][sensor:093]: 'ant-bms charge mosfet status code': Sending state 1.00000  with 0 decimals of accuracy
[16:23:54][D][text_sensor:064]: 'ant-bms charge mosfet status': Sending state 'On'
[16:23:54][D][sensor:093]: 'ant-bms discharge mosfet status code': Sending state 1.00000  with 0 decimals of accuracy
[16:23:54][D][text_sensor:064]: 'ant-bms discharge mosfet status': Sending state 'On'
[16:23:54][D][sensor:093]: 'ant-bms balancer status code': Sending state 0.00000  with 0 decimals of accuracy
[16:23:54][D][text_sensor:064]: 'ant-bms balancer status': Sending state 'Off'
[16:23:54][D][sensor:093]: 'ant-bms power': Sending state -2.73000 W with 2 decimals of accuracy
[16:23:54][D][sensor:093]: 'ant-bms max voltage cell': Sending state 8.00000  with 0 decimals of accuracy
[16:23:54][D][sensor:093]: 'ant-bms max cell voltage': Sending state 3.43500 V with 3 decimals of accuracy
[16:23:54][D][sensor:093]: 'ant-bms min voltage cell': Sending state 3.00000  with 0 decimals of accuracy
[16:23:54][D][sensor:093]: 'ant-bms min cell voltage': Sending state 3.39900 V with 3 decimals of accuracy
[16:23:54][D][sensor:093]: '{name} delta cell voltage': Sending state 0.03600 V with 3 decimals of accuracy
[16:23:54][D][sensor:093]: 'ant-bms average cell voltage': Sending state 3.41800 V with 2 decimals of accuracy
[16:23:54][D][sensor:093]: 'ant-bms battery strings': Sending state 8.00000  with 0 decimals of accuracy
```

Does this look like your battery back? ;-)

furang, 2023-06-19:
Thanks. Sorry for headache.

So I need to use **ant_bms_old_ble** platform?

syssi, 2023-06-19:
Do you want to retrieve the measurements via UART or BLE?

furang, 2023-06-19:
Doesn't matter, but I have only esp8266 (as for now) and afaik ble is not available.

syssi, 2023-06-19:
Correct. If you want to stick to the ESP8266 you have to use UART (`ant_bms` component). If you order a ESP32 the `ant_bms_old_ble` is the correct choice for BLE communication.

furang, 2023-06-19:
But I tried ant_bms (example at the beginning) and it didn't work. Also tried with display disconnected and the same.
Or mb should I change my config somehow, e.g. remove 
```yaml
    supports_new_commands: true

```

syssi, 2023-06-19:
There are no write instructions/ commands in your capture so we cannot be sure about the supported command set. The `ant_bms` component should work out of the box. Please make sure to connect `3.3V` and `GND` to the BMS because the BMS doesn't respond if the 3.3V line isn't pulled high.

furang, 2023-06-21:
Everything works fine with converter. If I attach esp I get the same messages with NaNs. Led blinks so I think wiring is fine. What else can I try?

syssi, 2023-06-21:
It looks like I've found the root cause. `D4` doesn't mean `GPIO4`. Please attach the RX wire to `D2`. This should work out of the box.

https://i0.wp.com/edistechlab.com/wp-content/uploads/2021/04/WeMos-d1-mini-Pin-out.png

furang, 2023-06-21:
OMFG. That works. It was not obvious.
Sorry one more time for wasting your time.
Now I faced another problem - it interferes with the display and display drops its values and then restores and so one. Have to solve this. But it has nothing in common with this ticket. It can be closed.
Thanks for your help and for your work and sorry for wasting your time.

syssi, 2023-06-22:
No problem. You are welcome! Please create a new issue if something doesn't work as expected.

closed 2023-06-22 by syssi