syssi/esphome-ant-bms #70: Interface with ANT 32S 300A - Tested

JorgeTorres01, 2023-07-01:
Hello,

I would like to leave only my test with the **ANT 32S 300A** .

I had some difficulty communicating with the BMS.
Through another topic I noticed that pins A, B is not RS485
I tested just connecting BMS to FTDI. 

[image]

[image]

syssi, 2023-07-01:
It looks like your device is compatible with this implementation. Just give it a try! :-)

syssi, 2023-07-16:
Did you try to change some settings (f.e. toggle the Mosfets) using this project? Is the old or new command set required?

closed 2023-07-16 by syssi

cng3035, 2023-08-21:
@JorgeTorres01
I have a ANT BMS 32S 200A that appears to have a very similar PCB as your 300A (same data pin layout).

I tried providing 3.3V to the V port but don't see any data on the "A" and "B" pin even with an oscilloscope.

Can you provide more details on how you hooked it up?
I'm using a generic "FT232RL" board and hooking up as follows

- VCC (in 3.3V mode) to the V pin on the bms
- GND to the G pin on the bms
- RX to A on the bms
- TX to B on the bms

pwilkowski, 2024-04-02:
@cng3035 have you solved the issues?

cng3035, 2024-04-02:
@pwilkowski Unfortunately no, was not able to get a wired serial connection working. Still using ant-bms-old-ble

commit d489cd21 (2023-07-16): Add ANT 32S 300A as supported (Closes: #70)