syssi/esphome-ant-bms #84: Bluetooth does not connect to ANT BMS

Deadobiwan, 2024-02-28:
Hello,

First of all, thanks for your project.

I tried to use it through bluetooth (bt) but it does not connect. I assume it is due to bt password. Indeed, when I try to connect with my PC to the ANT BMS a BT password/code is required the first time. In the log below, you can see the programm detect the bms bt but can not connect

[logs_bms2_run.txt](https://github.com/syssi/esphome-ant-bms/files/14430097/logs_bms2_run.txt)

Regards,

syssi, 2024-02-28:
There is a ANT-BMS reported but with a different MAC address:

```
[08:04:57][I][ble_adv:074]: New ANT-BMS found
[08:04:57][I][ble_adv:075]:   Name: ANT-BLE24BHUB-5300
[08:04:57][I][ble_adv:076]:   MAC address: 24:E2:4F:A7:53:00
[08:04:57][D][ble_adv:077]:   Advertised service UUIDs:
[08:04:57][D][ble_adv:079]:     - 0xFFE0
[08:04:57][D][ble_adv:079]:     - 0xFEE7
```

You are trying to connect to `24:E2:4F:A6:53:02`. Do you own multiple devices with different MAC addresses? If not please try to connect to `24:E2:4F:A7:53:00` instead.

Deadobiwan, 2024-02-29:
Hello,

Indeed, I have 2 ANT BMS one 24:E2:4F:A7:53:00 and one 24:E2:4F:A7:53:02. In the below log I tried to activate only the 02.

This is the logs with the old bt ant bms: [logs_bms2_logs.txt](https://github.com/syssi/esphome-ant-bms/files/14444154/logs_bms2_logs.txt)

This is the logs with the new ant bms: [logs_bms2_run(3).txt](https://github.com/syssi/esphome-ant-bms/files/14444223/logs_bms2_run.3.txt)

syssi, 2024-02-29:
That's pretty strange because the `ble_client` component is unable to establish a BLE client connection which is unusal. Do you have access to another ESP32 module just for testing? I'm worried because of the `S3` flavour of your ESP.

Deadobiwan, 2024-03-05:
Yes I have test with a start ESP it is the same behavior.

syssi, 2024-03-06:
Do you own an Android device and could you try to provide a btsnoop capture of the traffic between the Android App and your BMS?

Deadobiwan, 2024-03-06:
Unfortunately I don’t have Android devices. Do you have another solution? Otherwise I will have to ask around me.

syssi, 2024-03-07:
This is the only way I'm aware of to prepare a capture of the bluetooth traffic.

syssi, 2024-04-09:
@dead-obiwan Did you open the BMS app while recording the capture? I can see your Motorola device but there is no communication with a second device at the captures.

dead-obiwan, 2024-04-09:
Yes I did but I will retry tonight.

dead-obiwan, 2024-04-10:
Hello,

It should be good now

[hci_snoop20240410004640.cfa.log](https://github.com/syssi/esphome-ant-bms/files/14933319/hci_snoop20240410004640.cfa.log)

syssi, 2024-04-13:
Good job!

Write instructions / Commands extracted from the capture:

```
7e:a1:01:00:00:be:18:55:aa:55
7e:a1:02:6c:02:20:58:c4:aa:55
```

Some response examples extracted from the capture:

```
7e:a1:12:6c:02:20:32:34:42:48:00:00:00:00:00:00:00:00:00:00:00:00:32:34:42:48:55:42:30:30:2d:32:31:31:30:32:36:41:57:96:ff:0b:00:00:41:f2:aa:55
7e:a1:11:00:00:9e:05:01:04:16:00:00:00:00:00:00:00:00:00:30:80:21:00:00:00:00:00:00:00:00:00:00:00:00:df:08:ae:08:d7:08:cd:08:a6:08:a6:08:a6:08:a6:08:70:08:70:08:70:08:a8:08:88:08:a9:08:a6:08:a6:08:a6:08:a8:08:a6:08:a6:08:a6:08:d7:08:19:00:19:00:1a:00:19:00:1c:00:22:00:08:13:00:00:00:00:64:00:01:01:04:00:00:b4:c4:04:5d:2e:13:00:96:ac:db:00:00:00:00:00:e0:a4:a6:03:ff:e8:3f:00:df:08:01:00:70:08:09:00:6f:00:a6:08:00:00:6c:00:69:00:af:02:f3:fa:d2:52:d8:00:5a:06:df:00:e7:9e:5b:00:4a:10:8d:00:a3:c6:aa:55
7e:a1:11:00:00:9e:05:01:04:16:00:00:00:00:00:00:00:00:00:30:80:21:00:00:00:00:00:00:00:00:00:00:00:00:df:08:ae:08:d7:08:cd:08:a6:08:a6:08:a6:08:a6:08:70:08:70:08:70:08:a8:08:88:08:a9:08:a6:08:a6:08:a6:08:a8:08:a6:08:a6:08:a6:08:d7:08:19:00:19:00:1a:00:19:00:1c:00:22:00:08:13:00:00:00:00:64:00:01:01:04:00:00:b4:c4:04:5d:2e:13:00:96:ac:db:00:00:00:00:00:e0:a4:a6:03:ff:e8:3f:00:df:08:01:00:70:08:09:00:6f:00:a6:08:00:00:6c:00:69:00:af:02:f3:fa:d2:52:d8:00:5a:06:df:00:e7:9e:5b:00:4a:10:8d:00:a3:c6:aa:55
7e:a1:11:00:00:9e:05:01:04:16:00:00:00:00:00:00:00:00:00:30:80:21:00:00:00:00:00:00:00:00:00:00:00:00:df:08:ae:08:d7:08:cd:08:a6:08:a6:08:a6:08:a6:08:70:08:70:08:70:08:a8:08:88:08:a9:08:a6:08:a6:08:a6:08:a8:08:a6:08:a6:08:a6:08:d7:08:19:00:19:00:1a:00:19:00:1c:00:22:00:08:13:00:00:00:00:64:00:01:01:04:00:00:b4:c4:04:5d:2e:13:00:96:ac:db:00:00:00:00:00:e1:a4:a6:03:ff:e8:3f:00:df:08:01:00:70:08:09:00:6f:00:a6:08:00:00:6c:00:69:00:af:02:f3:fa:d2:52:d8:00:5a:06:df:00:e7:9e:5b:00:4a:10:8d:00:32:6a:aa:55
7e:a1:11:00:00:9e:05:01:04:16:00:00:00:00:00:00:00:00:00:30:80:21:00:00:00:00:00:00:00:00:00:00:00:00:df:08:ae:08:d7:08:cd:08:a6:08:a6:08:a6:08:a6:08:70:08:70:08:70:08:a8:08:88:08:a9:08:a6:08:a6:08:a6:08:a8:08:a6:08:a6:08:a6:08:d7:08:19:00:19:00:1a:00:19:00:1c:00:22:00:08:13:00:00:00:00:64:00:01:01:04:00:00:b4:c4:04:5d:2e:13:00:96:ac:db:00:00:00:00:00:e3:a4:a6:03:ff:e8:3f:00:df:08:01:00:70:08:09:00:6f:00:a6:08:ff:ff:6c:00:69:00:af:02:f3:fa:d2:52:d8:00:5a:06:df:00:e7:9e:5b:00:4a:10:8d:00:71:95:aa:55
7e:a1:11:00:00:9e:05:01:04:16:00:00:00:00:00:00:00:00:00:00:80:01:00:00:00:00:00:00:00:00:00:00:00:00:a6:08:ae:08:a7:08:a6:08:ae:08:ac:08:ad:08:ac:08:ad:08:a5:08:aa:08:a6:08:ac:08:a7:08:a9:08:ac:08:9c:08:a1:08:ac:08:ae:08:ac:08:aa:08:16:00:16:00:16:00:17:00:18:00:17:00:0d:13:02:00:5b:00:64:00:01:01:00:00:b8:ec:7e:04:9b:27:07:04:4f:4e:d4:00:09:00:00:00:ac:61:e8:02:00:00:00:00:ae:08:02:00:9c:08:11:00:12:00:a9:08:00:00:6c:00:67:00:ad:02:f3:fa:3a:f8:d0:00:64:a4:d7:00:5b:1c:5f:00:4c:62:99:00:dd:a8:aa:55
7e:a1:11:00:00:9e:05:01:04:16:00:00:00:00:00:00:00:00:00:00:80:01:00:00:00:00:00:00:00:00:00:00:00:00:a6:08:ae:08:a7:08:a6:08:ae:08:ac:08:ad:08:ac:08:ad:08:a5:08:aa:08:a6:08:ac:08:a7:08:a9:08:ac:08:9b:08:a1:08:ab:08:ae:08:ac:08:aa:08:16:00:16:00:16:00:17:00:18:00:17:00:0d:13:02:00:5b:00:64:00:01:01:00:00:b8:ec:7e:04:4e:25:07:04:4f:4e:d4:00:09:00:00:00:e5:61:e8:02:00:00:00:00:ae:08:02:00:9b:08:11:00:13:00:a9:08:00:00:6c:00:68:00:ad:02:f3:fa:3a:f8:d0:00:64:a4:d7:00:5b:1c:5f:00:4c:62:99:00:b2:0f:aa:55
7e:a1:11:00:00:9e:05:01:04:16:00:00:00:00:00:00:00:00:00:00:80:01:00:00:00:00:00:00:00:00:00:00:00:00:a6:08:ae:08:a7:08:a6:08:ae:08:ac:08:ad:08:ac:08:ad:08:a5:08:aa:08:a6:08:ac:08:a7:08:a9:08:ac:08:9b:08:a1:08:ab:08:ae:08:ac:08:aa:08:16:00:16:00:16:00:17:00:18:00:17:00:0d:13:02:00:5b:00:64:00:01:01:00:00:b8:ec:7e:04:4e:25:07:04:4f:4e:d4:00:09:00:00:00:e5:61:e8:02:00:00:00:00:ae:08:02:00:9b:08:11:00:13:00:a9:08:00:00:6c:00:68:00:ad:02:f3:fa:3a:f8:d0:00:64:a4:d7:00:5b:1c:5f:00:4c:62:99:00:b2:0f:aa:55
7e:a1:11:00:00:9e:05:01:04:16:00:00:00:00:00:00:00:00:00:00:80:01:00:00:00:00:00:00:00:00:00:00:00:00:a6:08:ae:08:a7:08:a6:08:ae:08:ac:08:ad:08:ac:08:ad:08:a5:08:aa:08:a6:08:ac:08:a7:08:a9:08:ac:08:9b:08:a1:08:ab:08:ae:08:ac:08:aa:08:16:00:16:00:16:00:17:00:18:00:17:00:0d:13:02:00:5b:00:64:00:01:01:00:00:b8:ec:7e:04:4e:25:07:04:4f:4e:d4:00:09:00:00:00:e6:61:e8:02:00:00:00:00:ae:08:02:00:9b:08:11:00:13:00:a9:08:00:00:6c:00:68:00:ad:02:f3:fa:3a:f8:d0:00:64:a4:d7:00:5b:1c:5f:00:4c:62:99:00:02:ba:aa:55
7e:a1:11:00:00:9e:05:01:04:16:00:00:00:00:00:00:00:00:00:30:80:21:00:00:00:00:00:00:00:00:00:00:00:00:df:08:ad:08:d7:08:cd:08:a6:08:a6:08:a6:08:a6:08:70:08:70:08:70:08:a7:08:88:08:a9:08:a6:08:a6:08:a6:08:a8:08:a6:08:a6:08:a6:08:d7:08:19:00:19:00:1a:00:19:00:1c:00:22:00:08:13:00:00:00:00:64:00:01:01:04:00:00:b4:c4:04:5d:2e:13:00:96:ac:db:00:00:00:00:00:6a:a5:a6:03:ff:e8:3f:00:df:08:01:00:70:08:09:00:6f:00:a6:08:00:00:6c:00:6a:00:af:02:f3:fa:d2:52:d8:00:5a:06:df:00:e7:9e:5b:00:4a:10:8d:00:04:fb:aa:55
7e:a1:12:6c:02:20:32:34:42:48:41:30:54:42:31:31:30:41:00:00:00:00:32:34:42:48:55:42:30:30:2d:32:32:30:38:32:33:41:f0:9c:ff:0b:00:00:41:f2:aa:55
7e:a1:11:00:00:9e:05:01:04:16:00:00:00:00:00:00:00:00:00:00:80:01:00:00:00:00:00:00:00:00:00:00:00:00:a6:08:ae:08:a7:08:a6:08:ae:08:ac:08:ad:08:ac:08:ad:08:a4:08:aa:08:a5:08:ac:08:a7:08:a9:08:ac:08:9b:08:a1:08:ab:08:ae:08:ac:08:aa:08:16:00:16:00:16:00:17:00:18:00:17:00:0d:13:02:00:5b:00:64:00:01:01:00:00:b8:ec:7e:04:ae:22:07:04:4f:4e:d4:00:09:00:00:00:24:62:e8:02:00:00:00:00:ae:08:02:00:9b:08:11:00:13:00:a9:08:00:00:6c:00:67:00:ad:02:f3:fa:3a:f8:d0:00:64:a4:d7:00:5b:1c:5f:00:4 [... cut ...]
7e:a1:11:00:00:9e:05:01:04:16:00:00:00:00:00:00:00:00:00:00:80:01:00:00:00:00:00:00:00:00:00:00:00:00:a6:08:ae:08:a7:08:a6:08:ae:08:ac:08:ad:08:ac:08:ad:08:a4:08:aa:08:a5:08:ac:08:a7:08:a9:08:ac:08:9b:08:a1:08:ab:08:ae:08:ac:08:aa:08:16:00:16:00:16:00:17:00:18:00:17:00:0d:13:02:00:5b:00:64:00:01:01:00:00:b8:ec:7e:04:ae:22:07:04:4f:4e:d4:00:09:00:00:00:24:62:e8:02:00:00:00:00:ae:08:02:00:9b:08:11:00:13:00:a9:08:00:00:6c:00:68:00:ad:02:f3:fa:3a:f8:d0:00:64:a4:d7:00:5b:1c:5f:00:4c:62:99:00:21:8d:aa:55
```
[traffic.txt](https://github.com/syssi/esphome-ant-bms/files/14967935/traffic.txt)

Deadobiwan, 2024-04-13:
Good, do you understand why it does not work?

syssi, 2024-04-13:
Good news: The payload can be decoded in general by the current implementation:

```
[18:08:36][I][ant_bms_ble:232]: Status frame (168 bytes):
[18:08:37][D][ant_bms_ble:233]:   7E.A1.11.00.00.9E.05.01.04.16.00.00.00.00.00.00.00.00.00.30.80.21.00.00.00.00.00.00.00.00.00.00.00.00.DF.08.AE.08.D7.08.CD.08.A6.08.A6.08.A6.08.A6.08.70.08.70.08.70.08.A8.08.88.08.A9.08.A6.08.A6.08.A6.08.A8.08.A6.08.A6.08.A6.08.D7.08.19.00.19.00.1A.00.19.00.1C.00.22.00.08.13.00.00.00.00.64.00.01.01.04.00.00.B4.C4.04.5D.2E.13.00.96.AC.DB.00.00.00.00.00.E0.A4.A6.03.FF.E8.3F.00.DF.08.01.00.70.08.09.00.6F.00.A6.08.00.00.6C.00.69.00.AF.02.F3.FA.D2.52.D8.00.5A.06.DF.00.E7.9E.5B.00.
[18:08:37][I][ant_bms_ble:251]:   Permissions: 5
[18:08:37][I][ant_bms_ble:254]:   Battery status: 1
[18:08:37][I][ant_bms_ble:258]:   Number of temperature sensors: 4
[18:08:37][D][sensor:094]: 'ant-bms-ble battery strings': Sending state 22.00000  with 0 decimals of accuracy
[18:08:37][D][sensor:094]: 'ant-bms-ble cell voltage 1': Sending state 2.27100 V with 3 decimals of accuracy
[18:08:37][D][sensor:094]: 'ant-bms-ble cell voltage 2': Sending state 2.22200 V with 3 decimals of accuracy
[18:08:37][D][sensor:094]: 'ant-bms-ble cell voltage 3': Sending state 2.26300 V with 3 decimals of accuracy
[18:08:37][D][sensor:094]: 'ant-bms-ble cell voltage 4': Sending state 2.25300 V with 3 decimals of accuracy
[18:08:37][D][sensor:094]: 'ant-bms-ble cell voltage 5': Sending state 2.21400 V with 3 decimals of accuracy
[18:08:37][D][sensor:094]: 'ant-bms-ble cell voltage 6': Sending state 2.21400 V with 3 decimals of accuracy
[18:08:37][D][sensor:094]: 'ant-bms-ble cell voltage 7': Sending state 2.21400 V with 3 decimals of accuracy
[18:08:37][D][sensor:094]: 'ant-bms-ble cell voltage 8': Sending state 2.21400 V with 3 decimals of accuracy
[18:08:37][D][sensor:094]: 'ant-bms-ble cell voltage 9': Sending state 2.16000 V with 3 decimals of accuracy
[18:08:37][D][sensor:094]: 'ant-bms-ble cell voltage 10': Sending state 2.16000 V with 3 decimals of accuracy
[18:08:37][D][sensor:094]: 'ant-bms-ble cell voltage 11': Sending state 2.16000 V with 3 decimals of accuracy
[18:08:37][D][sensor:094]: 'ant-bms-ble cell voltage 12': Sending state 2.21600 V with 3 decimals of accuracy
[18:08:37][D][sensor:094]: 'ant-bms-ble cell voltage 13': Sending state 2.18400 V with 3 decimals of accuracy
[18:08:37][D][sensor:094]: 'ant-bms-ble cell voltage 14': Sending state 2.21700 V with 3 decimals of accuracy
[18:08:37][D][sensor:094]: 'ant-bms-ble cell voltage 15': Sending state 2.21400 V with 3 decimals of accuracy
[18:08:37][D][sensor:094]: 'ant-bms-ble cell voltage 16': Sending state 2.21400 V with 3 decimals of accuracy
[18:08:37][D][sensor:094]: 'ant-bms-ble cell voltage 17': Sending state 2.21400 V with 3 decimals of accuracy
[18:08:37][D][sensor:094]: 'ant-bms-ble cell voltage 18': Sending state 2.21600 V with 3 decimals of accuracy
[18:08:37][D][sensor:094]: 'ant-bms-ble cell voltage 19': Sending state 2.21400 V with 3 decimals of accuracy
[18:08:37][D][sensor:094]: 'ant-bms-ble cell voltage 20': Sending state 2.21400 V with 3 decimals of accuracy
[18:08:37][D][sensor:094]: 'ant-bms-ble cell voltage 21': Sending state 2.21400 V with 3 decimals of accuracy
[18:08:37][D][sensor:094]: 'ant-bms-ble cell voltage 22': Sending state 2.26300 V with 3 decimals of accuracy
[18:08:37][D][sensor:094]: 'ant-bms-ble temperature 1': Sending state 25.00000 °C with 0 decimals of accuracy
[18:08:37][D][sensor:094]: 'ant-bms-ble temperature 2': Sending state 25.00000 °C with 0 decimals of accuracy
[18:08:37][D][sensor:094]: 'ant-bms-ble temperature 3': Sending state 26.00000 °C with 0 decimals of accuracy
[18:08:37][D][sensor:094]: 'ant-bms-ble temperature 4': Sending state 25.00000 °C with 0 decimals of accuracy
[18:08:37][D][sensor:094]: 'ant-bms-ble temperature 5': Sending state 28.00000 °C with 0 decimals of accuracy
[18:08:37][D][sensor:094]: 'ant-bms-ble temperature 6': Sending state 34.00000 °C with 0 decimals of accuracy
[18:08:37][D][sensor:094]: 'ant-bms-ble total voltage': Sending state 48.72000 V with 2 decimals of accuracy
[18:08:37][D][sensor:094]: 'ant-bms-ble current': Sending state 0.00000 A with 1 decimals of accuracy
[18:08:37][D][sensor:094]: 'ant-bms-ble soc': Sending state 0.00000 % with 0 decimals of accuracy
[18:08:37][I][ant_bms_ble:319]:   State of health: 100 %
[18:08:37][D][sensor:094]: 'ant-bms-ble charge mosfet status code': Sending state 1.00000  with 0 decimals of accuracy
[18:08:37][V][text_sensor:013]: 'ant-bms-ble charge mosfet status': Received new state On
[18:08:37][D][text_sensor:064]: 'ant-bms-ble charge mosfet status': Sending state 'On'
[18:08:37][D][sensor:094]: 'ant-bms-ble discharge mosfet status code': Sending state 1.00000  with 0 decimals of accuracy
[18:08:37][V][text_sensor:013]: 'ant-bms-ble discharge mosfet status': Received new state On
[18:08:37][D][text_sensor:064]: 'ant-bms-ble discharge mosfet status': Sending state 'On'
[18:08:37][D][sensor:094]: 'ant-bms-ble balancer status code': Sending state 4.00000  with 0 decimals of accuracy
[18:08:37][V][text_sensor:013]: 'ant-bms-ble balancer status': Received new state Automatic equalization
[18:08:37][D][text_sensor:064]: 'ant-bms-ble balancer status': Sending state 'Automatic equalization'
[18:08:37][D][sensor:094]: 'ant-bms-ble total battery capacity setting': Sending state 80.00000 Ah with 2 decimals of accuracy
[... cut ...]
[18:09:22][D][ant_bms_ble:417]:   7E.A1.12.6C.02.20.32.34.42.48.00.00.00.00.00.00.00.00.00.00.00.00.32.34.42.48.55.42.30.30.2D.32.31.31.30.32.36.41.57.96.FF.0B.00.00.41.F2.AA.55 (48)
[18:09:22][V][text_sensor:013]: 'ant-bms-ble device model': Received new state 24BH
[18:09:22][D][text_sensor:064]: 'ant-bms-ble device model': Sending state '24BH'
[18:09:22][V][text_sensor:013]: 'ant-bms-ble software version': Received new state 24BHUB00-211026A
[18:09:22][D][text_sensor:064]: 'ant-bms-ble software version': Sending state '24BHUB00-211026A'
``` 

I will have a look at the BLE client connection setup now. I guess there is something different to the past models.

syssi, 2024-04-13:
Details about the connection setup:

```
1. Write Request

Bluetooth Attribute Protocol
    Opcode: Write Request (0x12)
    Handle: 0x0011 (Unknown: Unknown: Client Characteristic Configuration)
    Characteristic Configuration Client: 0x0001, Notification
    
# Response:
# 
# Bluetooth Attribute Protocol
#     Opcode: Write Response (0x13)
#     [Handle: 0x0011 (Unknown: Unknown: Client Characteristic Configuration)]
#     [Request in Frame: 11289]

2. Write Command

Bluetooth Attribute Protocol
    Opcode: Write Command (0x52)
    Handle: 0x0010 (Unknown: Unknown)
    Value: 7ea1026c022058c4aa55

# No response

3. Write Command

Bluetooth Attribute Protocol
    Opcode: Write Command (0x52)
    Handle: 0x0010 (Unknown: Unknown)
    Value: 7ea1010000be1855aa55

# Incoming notifications on handle 0x0010

5. Write Command

Bluetooth Attribute Protocol
    Opcode: Write Command (0x52)
    Handle: 0x0010 (Unknown: Unknown)
    Value: 7ea1010000be1855aa55

# Incoming notifications on handle 0x0010

6. Write Command

Bluetooth Attribute Protocol
    Opcode: Write Command (0x52)
    Handle: 0x0010 (Unknown: Unknown)
    Value: 7ea1010000be1855aa55

# Incoming notifications on handle 0x0010
    
7. Write Command

Bluetooth Attribute Protocol
    Opcode: Write Command (0x52)
    Handle: 0x0010 (Unknown: Unknown)
    Value: 7ea1010000be1855aa55

# Incoming notifications on handle 0x0010
```

syssi, 2024-04-13:
Hmm... please flash this YAML another time: https://github.com/syssi/esphome-ant-bms/blob/main/esp32-ble-example.yaml

And replace the `logger` section with:

```
logger:
  level: VERY_VERBOSE
  logs:
    ant_bms_ble: VERY_VERBOSE
    ble_client: DEBUG
    esp32_ble: DEBUG
    esp32_ble_tracker: DEBUG
    ota: INFO
    wifi: INFO
    scheduler: DEBUG
    component: DEBUG
    sensor: DEBUG
    mqtt: INFO
    mqtt.idf: INFO
    mqtt.component: INFO
    mqtt.sensor: INFO
    mqtt.switch: INFO
    api.service: INFO
    api: INFO
```

In general your device should be supported out of the box. Please make sure no other BLE client blocks the single supported BLE connection. Please provide another log.

dead-obiwan, 2024-04-14:
Hello,

Here is the file log. No other BL connection were initiated. I have change the YAML accordingly to your previous post. The software was setup to connect to BMS-ANT24BHUB-5300 mac_address: 24:E2:4F:A6:53:00.

[2024-04_14-logs_bms2_run.txt](https://github.com/syssi/esphome-ant-bms/files/14970102/2024-04_14-logs_bms2_run.txt)

syssi, 2024-04-15:
There is a typo in your MAC address:

```
Device address:    24:E2:4F:A7:53:00
Used at the YAML:  24:E2:4F:A6:53:00
                            ^^
```

Please update your YAML an try again.

dead-obiwan, 2024-04-15:
Indeed, shame on me. Is it possible to exchange in private message?

syssi, 2024-04-15:
Feel free to contact me via mail: basti@linkt.de

syssi, 2024-04-15:
It's working now.

closed 2024-04-15 by syssi