syssi/esphome-ant-bms #87: Very old antbms (< 2019) not communating

SeByDocKy, 2024-04-15:
Hi @syssi,

I tried both the old and ant_bms platform without success. I got an old custom components working with : https://pastebin.com/0bBPx33L

Maybe you can see the difference with your current implementations ....

syssi, 2024-04-15:
I've extracted some details from the code posted to pastebin:

```

19200 baud

Periodic request: 5A.5A.00.00.01.01

Response:

0  0xAA
1  0x55
2  0xAA
3  0xFF
4         total voltage
5         total voltage
6         cell1 voltage
7         cell1 voltage
8         cell2 voltage
9         cell2 voltage
10        cell3 voltage
11        cell3 voltage
12        cell4 voltage
13        cell4 voltage
...
34        cell15 voltage
35        cell15 voltage
36        cell16 voltage
37        cell16 voltage
...
70        current
71        current
72        current
73        current
74        state of charge
75        total capacity
76        total capacity
77        total capacity
78        total capacity
79        remaining capacity
80        remaining capacity
81        remaining capacity
82        remaining capacity
83        capacity
84        capacity
85        capacity
86        capacity
91        t_fet
92        t_fet
93        t_bal
94        t_bal
95        temp1
96        temp1
97        temp2
98        temp2
99        temp3
100       temp3
101       temp4
102       temp4
103       charge_idx
104       discharge_idx
105       balance_idx
...
111       puiss
112       puiss
113       puiss
114       puiss
...
116       max cell voltage
117       max cell voltage
118
119       min cell voltage
120       min cell voltage
121       average cell voltage
122       average cell voltage
123       cell count
132       cells_balanced_idx
135?      cells_balanced_idx
```

The status frame looks compatible to this implementation: https://github.com/syssi/esphome-ant-bms/blob/main/components/ant_bms_old/ant_bms_old.cpp#L177-L286

syssi, 2024-04-16:
Please try this YAML configuration: https://github.com/syssi/esphome-ant-bms/blob/main/esp8266-old-example.yaml

And provide some logs + photos of your wiring.

SeByDocKy, 2024-04-16:
Yes it should work.... with the old protocol.... Is the RX buffer size set to 384 a must have ?

syssi, 2024-04-16:
A complete response frame must fit into the buffer. These frames have a length of 140 bytes. A RX buffer of 150 bytes should work too probably!

closed 2024-04-16 by syssi