syssi/esphome-ant-bms #90: Invalid response fro BMS

Rudolf-Ja, 2024-06-06:
Hi,
I have problem with response from BMS. I have ANT 10S-24S. I use pins GPIO0 for Tx and GPIO2 for Rx, but tried GPIO4/5 and 1/3 too.
I use ESP8266 (Wemos D1 mini).
In LOG is "Invalid header" written (Line108). I tried to expand LOG message and values for first 4 elements of array "raw" were 0x00, 0x00, 0x00, 0x00.
I tried to meassure traffic on transmition PIN with osciloscope and there was: 0x5a, 0x96, 0xa0, 0x08, 0x02, 0x80, 0xa0. I expected 0x5a, 0x5a, 0x00, 0x00, 0x01, 0x01 as in the read_registers method written is. I don't understand why.
I tried esp8266-example.yaml and esp8266-old-example.yaml - the same result.

Please, do you have any idea, what could be wrong?

Thank you

syssi, 2024-06-06:
Could you enable the `debug` mode of the `uart` component and provide some logs?

```
logger:
  level: DEBUG

uart:
  - id: uart_0
    baud_rate: 19200
    rx_buffer_size: 384
    tx_pin: ${tx_pin}
    rx_pin: ${rx_pin}
    debug:
      direction: BOTH
```

This setting should print the outgoing RAW traffic into the log.

Rudolf-Ja, 2024-06-07:
Here is the LOG.
Whole line "Read_registers" and numbers in line "Invalid header" were added by me. First four numbers are raw[0] ... raw[3], fifth is the size of rx_buffer.
[image]

Rudolf-Ja, 2024-06-09:
The problem is probably already solved. There were three issues:
1) A mistake in decoding of oscilloscope data - the packet in my first post was incorrectly recognized and in reality is packet OK (5A 5A 00 00 ...) - some problem with start and stop bits interpretation.
2) Conector on side of BMS display has 6 pins. There are GND, battery +, GND, Rx, Tx, 3.3V. Only first GND is connected on true ground of battery. The second GND is only connected in the display. When I unplug the display and connect ESP instead, only the first PIN has the correct GND. In my wiring are both GND PIN connected between.
2) Difference between voltage level 0 and 1 on transmitting from the BMS is only 1.16V. Recognizing treshold of ESP was outside of the range between 0 and 1 on BMS transmition. I used a diode and a resistor to move the voltage level higher (thanks to my friend for the advice).

Thanks a lot for your support, syssi!

syssi, 2024-06-09:
Could you provide some additional details / schematics of your quirk mentioned at number 3? I remember some more users having trouble to talk to the BMS.

closed 2024-06-09 by syssi

Rudolf-Ja, 2024-06-09:
Here is my wiring.
[image]

Rudolf-Ja, 2024-06-09:
The block in the middle is connector unpluged from display. I am connected on it.