syssi/esphome-ant-bms #91: length of value must be at least 8 but the old BMS only 4

jinkexue, 2024-06-20:
substitutions:
  name: ant-bms-old-ble
  device_description: "Monitor and control a ANT Battery Management System (ANT-BMS) via BLE"
  external_components_source: github://syssi/esphome-ant-bms@main
  # Apply the MAC address of your ANT-BMS here (BLE device name: "ANT-BLE24C"). Please don't use "BMS-ANT24C".
  mac_address: 16:aa:22:02:23:45
  password: "1234"

but feedback:

INFO Updating https://github.com/syssi/esphome-ant-bms.git@main
Failed config

ant_bms_old_ble: [source /config/esphome/ant32.yaml:96]
  - id: bms0
    ble_client_id: client0
    update_interval: 5s
    
    length of value must be at least 8.
    password: 1234

closed 2024-06-20 by syssi

jinkexue, 2024-06-20:
Failed config

ant_bms_old_ble: [source /config/esphome/ant32.yaml:97]
  - id: bms0
    ble_client_id: client0
    update_interval: 5s
    
    length of value must be at most 4.
    password: 12345678

could i change the bms password to  12345678  ?

syssi, 2024-06-20:
The `1234` password should be accepted now. Please give it a try!

jinkexue, 2024-06-20:
> The `1234` password should be accepted now. Please give it a try!

thanks, so quick 
at most 4 now ？
It seems I won’t be able to link to the BMS if I change the password to more than 4 digits

syssi, 2024-06-20:
Do you know which password lengths are supported in general?

cng3035, 2024-06-25:
> Do you know which password lengths are supported in general?

I currently have a 8 character password set and it was functional. The new change, `(min=4, max=4)`, is a breaking change for at least me and likely others.

I think an acceptable interim compromise would be
`cv.All(cv.string_strict, cv.Length(min=4, max=8)),`

syssi, 2024-06-25:
Thanks for your feedback! I will revert the change because a 4 digit password won't work either at the moment.

commit a8845b40 (2024-06-20): Fix required password length (Closes: #91)
commit 3d4a895d (2024-06-20): Fix required password length (Closes: #91) (#92)
commit 29e0b298 (2024-06-25): Revert "Fix required password length (Closes: #91) (#92)"

This reverts commit 3d4a895dbbeca262afeadaffb1e8717b897100f4.